apache/mynewt-nimble #887: Allow setting non-default IRK

jimmo, 2020-11-26:
hi! I'm from the [MicroPython](https://github.com/micropython/micropython) project, where we use NimBLE as our BLE stack (on top of a few different HCI controllers). I've recently been doing some work with pairing & bonding, and seem to be missing something important -- I can't see a way to not use the default IRK ble_hs_pvcy_default_irk.

ble_hs_pvcy_set_our_irk is in a private header (ble_hs_pvcy_priv.h), and even if it wasn't there's no way to avoid it already being called at startup (ble_hs_startup_go calls ble_hs_pvcy_set_our_irk(NULL))

But even if do use ble_hs_pvcy_set_our_irk is in a private header (ble_hs_pvcy_priv.h), and even if it wasn't there's no way to avoid it already being called at startup to re-set it in sync_cb or somewhere, then now all the peer IRK entries loaded from my key store will be thrown away, so I'd have to call ble_hs_misc_restore_irks to re-load them. But that's also in a private header.

I notice in #517 there's a reference to "setting a custom IRK". I'm not sure what the mechanism to set a custom IRK is, but in our case (MicroPython) this should ideally be able to be generated and set at runtime (i.e. not a syscfg option).

How are other NimBLE implementors using a non-default IRK in their devices? Thanks!

jimmo, 2022-08-25:
Hi NimBLE team, just wondering if you have any update on this. In MicroPython we continue to use a workaround similar to what I described above via the private functions. If other implementors are not doing this though, it is a big privacy issue to have all devices sharing the same IRK (as it allows tracking).

See also https://github.com/espressif/esp-nimble/issues/24

sjanc, 2022-11-03:
Hi,

https://github.com/apache/mynewt-nimble/pull/1396   is an initial step to handle this (at some point we should remove fixed IRK and just use random)