apache/mynewt-nimble #890: ble_gap_terminate timeout?

btsimonh, 2020-12-09:
Hi all, 
trying to track down a problem in esp32 arduino-nimble in Tasmota.

a 'normal' disconnect logs:
```
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
0x06 0x04 0x03 0x00 0x00 0x13
Command Status: status=0 cmd_pkts=5 ocf=0x6 ogf=0x1
D NimBLEClient: "ble_gap_terminate success: rc=0 "
D NimBLEClient: "<< disconnect()"
Disconnection Complete: status=0 handle=0 reason=22
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
```

but my BLE stops after a few minutes of connect/write/disconnect cycles with the below.
Seemingly, the disconnect confirmation never received, and there appears to be no timeout processing around the terminate procedure.
What is the correct approach to resolve this?  Should arduino-nimble look for a timeout, and kill the connection?  if so how?
Or is terminate timeout something which has been missed in mynewt?
```
ble_hs_hci_acl_tx(): 0x00 0x00 0x08 0x00 0x04 0x00 0x04 0x00 0x12 0x11 0x04 0x03
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=5 data=0x01 0x00 0x04 0x00 0x13 
rxed att command: write rsp; conn=0 
I NimBLERemoteCharacteristic: "Write complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "<< writeValue, rc: 0"
09:51:08 SL1-29 write characteristic
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=13 data=0x09 0x00 0x04 0x00 0x1b 0x21 0x04 0x02 0x01 0x08 0x00 0x04 0x2a 
rxed att command: notify req; conn=0 handle=0x0421
D NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 1057"
D NimBLEClient: "checking service 3e135142-654f-9090-134a-a6ff5bb77046 for handle: 1057"
D NimBLEClient: ">> disconnect()"
**GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19**
ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
0x06 0x04 0x03 0x00 0x00 0x13
Command Status: status=0 cmd_pkts=5 ocf=0x6 ogf=0x1
D NimBLEClient: "ble_gap_terminate success: rc=0 "
D NimBLEClient: "<< disconnect()"
09:51:09 SL2-28 operation complete
09:51:09 SL1-29 runTaskDoneOperation: disconnecting connected client
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /cs search: c2=244
.......
[V][Parsing.cpp:231] _parseRequest():  Arguments: c2=244
09:51:11 SL1-29 ############################################# wait disconnect 1
09:51:11 SL1-29 ############################################# wait disconnect 2
09:51:12 SL1-29 ############################################# wait disconnect 3
09:51:13 SL1-29 ############################################# wait disconnect 4
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /cs search: c2=244
```

rymanluk, 2020-12-17:
For me it looks like it is the controller not sending disconnection complete event.
Do you have HCI logs possibly?

btsimonh, 2020-12-17:
yep, much logging later.
The controller acknowledges the disconnect command, but we hear nothing after that.
I'm told by @h2zero that the controller should be responsible for the supervision timeout, and so if the dev does not send anything back, we should still see some action some time after asking for disconnect.
So..  the question is should NimBLE try to deal with such 'bad behaviour'?  It does have timeouts for some things, but this is not one of them currently.  I did try a mod implementing a timeout; but did not know how to correctly kill everything regarding the connection.  For the moment, I'm timing out on the outside and simply resetting the whole of BLE.
br,
simon

sjanc, 2022-04-13:
I don't think we can do anything reasonable here if controller is misbehaving...

closed 2022-04-13 by sjanc