apache/mynewt-nimble #893: le coded phy scanning on nRF52811

Soichiro-Iwataki, 2020-12-14:
Hello, 
I'm trying to use nRF52811 with apps/blehci as bluetooth controller which is connected via uart hci.
Scan on 1Mbps PHY is OK, but scan on coded phy has problem.
Here is detailed explanation.
- Environment:
  - mcu nrf52811
  - tool chain gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update) 
  - mynewt-core 	1.8.0
  - mynewt-nimble 1.3.0
  - mcu is connected to linux host (Linux 29b841b 4.19.118 armv6l) via uart.

  - syscfg setting
	`MCU_TARGET: nRF52811`
	`MCU_LFCLK_SOURCE: LFRC`
`BLE_LL_CFG_FEAT_LE_CODED_PHY: 1`
`BLE_EXT_ADV: 1`
    - syscfg.vals.BLE_CONTROLLER:
    `TIMER_0: 0`
    `TIMER_5: 1`
    `OS_CPUTIME_FREQ: 32768`
    `OS_CPUTIME_TIMER_NUM: 5`

 
- What I tried:
Following commands was tested with uart tx/rx is monitored by oscilloscope
  - `btattach -B /dev/ttyAMA0 -P h4 -S 1000000`
  - `bluetoothctl`
  - `scan on`
    - scan start with 1M phy and coded phy (1)
  - `scan off`
  - `hcitool cmd 08 31 03 04 04`
    - set default phy coded only
  - `bluetoothctl`
  - `scan on`
    - scan start with coded phy (2)
  - `scan off`
- Expected result:
  - scan result of (1) contains both of 1M phy devices and coded phy devices
  - scan result of (2) contains coded phy devices 
  - both of (1) and (2) is extended scan
- Actual result:
  - scan result of (1) contains only 1M phy devices
  - scan result of (2) contains no devices
  - both of (1) and (2) is extended scan

I think uart hci communication host and controller, extended scan ,and 1Mbps phy are ok.
I'm wondering if there is bug on coded phy, or I have incorrect configuration. 
Thanks.

sjanc, 2022-04-13:
is this stil an issue with latest version?

Soichiro-Iwataki, 2022-04-14:
@sjanc Thank you for comment.
I have not tested with latest version. 
If I get the chance I'd like to try with latest version.