apache/mynewt-nimble #909: No "Class of Device" (CoD) setter, getter found in API?

codewitch-honey-crisis, 2021-02-02:
I need to get the CoD so I can determine if the devices I'm finding are the specific devices I created.

Class of Device is how you do this, no?

bluedroid has this
BluetoothClass
and it gets me that information.

There is no equivalent here from the looks of things, either for setting it (server) or getting it (client)

I'm looking for a different lightweight library. What would you have me do, connect by name and manufacturer? I am uncomfortable with this without checking the device class.

Or maybe it is here, and the documentation doesn't make it clear.

Either way, it's a problem.

Best regards

andrzej-kaczmarek, 2021-02-02:
There's no CoD in BLE, it's only used in Bluetooth classic which NimBLE does not support.
If you want to filter devices during discovery you can use either Appearance value or Services UUIDs, both can be present in advertising data.

codewitch-honey-crisis, 2021-02-02:
I'm sorry. I'm still new to this, and I thought CoD was how you did it.
Thanks!

closed 2021-02-08 by sjanc