apache/mynewt-nimble #912: Controller: Assertion fail in ble_ll_scan.c with extended advertisings

janosbrodbeck, 2021-02-10:
While running experiments with extended advertisings I encountered a reproducable assertion fail in the `ble_ll_scan_aux_data_unref()` function. The exact position is: `ble_ll_scan.c:1092`. I could trigger this assert only in scenarios when a node scans and advertises extended advertisings. Only advertising or only scanning could not trigger this until now. 

Unfortunately, the error cannot be triggered on purpose, but occurs 'randomly' after a while. Sometimes after a few minutes/hours, sometimes all experiments run for hours without a problem. But higher payload and lower intervals seems to trigger it with a higher chance.

I've written a [small and simple example application](https://github.com/janosbrodbeck/nimble_app_assertion_fail) for mynewt to reproduce it. This application more or less simulates ICMPv6 pings with a payload of 1100 bytes. It needs two nodes (e.g. nrf52dk/nrf52840) to run. Brief instructions are in the repo. The error should be triggered very fast with an interval between 500ms and 300ms.

Some background:
* Originally I ran with RIOT into this issue but reproduced it with mynewt in the same scenarions
* Scan parameters are nimBLE (fast window/interval) defaults. Scan duration/window is set to 0 = continously scanning.
* Advertising happens with `FAST_INTERVAL1_MIN`/`FAST_INTERVAL1_MAX`, 1M PHY. 

This is the assertion:
```
# 017040 Assert @ 0x1a6d1 - repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_scan.c:1092
# 017040 Unhandled interrupt (2), exception sp 0x20004710
# 017040  r0:0x0000005e  r1:0x00000000  r2:0x80000000  r3:0xe000ed00
# 017040  r4:0x0001a6d1  r5:0x0002bb0c  r6:0x00000444  r7:0x00000000
# 017040  r8:0x000000f8  r9:0x00000001 r10:0x000000f8 r11:0x2000235c
# 017040 r12:0xa0000000  lr:0x0000c89f  pc:0x0000c88e psr:0x61000000
# 017040 ICSR:0x00421802 HFSR:0x00000000 CFSR:0x00000000
# 017040 BFAR:0xe000ed38 MMFAR:0xe000ed34
```

sjanc, 2021-02-15:
this appears to be assert from ble_ll_scan_aux_data_unref

BLE_LL_ASSERT(!(aux_data->flags_ll & BLE_LL_AUX_FLAG_HCI_SENT_ANY) ||
                      ((aux_data->flags_ll & BLE_LL_AUX_FLAG_HCI_SENT_ANY) &&
                       (aux_data->flags_ll & (BLE_LL_AUX_FLAG_HCI_SENT_COMPLETED | BLE_LL_AUX_FLAG_HCI_SENT_TRUNCATED))));

Could you show backtrace and try checking state of aux_data->flags_ll when this happens?
Or preferably, provide a core dump

janosbrodbeck, 2021-02-15:
Backtrace:
I've created the [backtrace as gist](https://gist.github.com/janosbrodbeck/6c2653a98449882ca8d8a1dc386d99f5). 

`aux_data->flags_ll` was set to decimal 226. So if I'm not mistaken:
`BLE_LL_AUX_FLAG_HCI_SENT_ANY`, `BLE_LL_AUX_FLAG_SCAN_ERROR`, `BLE_LL_AUX_FLAG_AUX_ADV_RECEIVED` and `BLE_LL_AUX_FLAG_AUX_CHAIN_RECEIVED` when it happens.

I was not able to get the core dump. I've set `OS_COREDUMP` and `IMGMGR_COREDUMP` to 1 in the target of the bootloader but could not get it with newtmgr. Created a connection profile for my board `nrf52840: type=serial, connstring='dev=/dev/ttyACM0'`. When trying to download the core dump with `newtmgr -c nrf52840 image coredownload <dl-location>` I just get nmp timeouts.

andrzej-kaczmarek, 2021-02-16:
@janosbrodbeck instead of coredump you can just dump RAM address area to binary file and attach it with application .elf. We will also try to reproduce this locally here with your app but it can take a while so if you can provide such dump we can try to analyze quickly.

janosbrodbeck, 2021-02-18:
I'm not sure if this is correct but here is my .elf file of the application and a dump of the memory (start: 0 end: 512000).
[memory_dump.zip](https://github.com/apache/mynewt-nimble/files/5999708/memory_dump.zip)

andrzej-kaczmarek, 2021-02-22:
@janosbrodbeck we reproduced this locally and found the issue, here's fix: https://github.com/apache/mynewt-nimble/pull/920. Seems to work after quick verification, we'll leave it overnight to be sure.

janosbrodbeck, 2021-02-22:
Nice, thank you! I start my boards and run it under heavy load as well to test it.

janosbrodbeck, 2021-02-23:
Experienced no crash overnight.

andrzej-kaczmarek, 2021-02-23:
@janosbrodbeck thanks, I've just merged the fix.

closed 2021-02-23 by andrzej-kaczmarek