apache/mynewt-nimble #926: How to make the btshell app support smp?

itewqq, 2021-03-01:
Hello, I have noticed that the comment in syscfg of btshell shows such a sentence

> SMP is not supported in this app, so disable smp-over-shell.

I want to know how to adjust btshell to support SMP?

Or just talk about my original needs:
My current requirement is to build a bonded and link-layer encrypted uart application. I made a simple modification to your ```bleuart``` app and make it supports this. But now I can only use the app on my mobile phone to connect to it. I originally wanted to use the api provided by btshell but found that it does not support SMP. Do you have any suggestions?

sjanc, 2021-03-02:
There is an unfortunate acronym collision in mynewt....  This comment is for SHELL_MGMT which is for mcumgr SMP (Simple Management Protocol) and not for BLE Security Manager.

BLE bonding is supported in btshell sample :-)

itewqq, 2021-03-02:
Get it. Thanks for the clarification.

closed 2021-03-03 by itewqq