apache/mynewt-nimble #940: blehci-usb app fails on assert when trying to upload an image using newtmgr

ncasaril, 2021-03-18:
Newtmgr stalls on image upload. 
Not entirely certain that this is an issue in nimble, as it may very well be within newtmgr also. However, newtmgr successfully uploads images using the same hw but with a different hci dongle / built in ble dev. 

To reproduce: 
- setup a pca10056 board / nrf52840-dongle with the blehci app using usb as transport. Should become a hci device when connected in linux. 
- using newtmgr under linux upload an image over ble using the created hci device

The below assert gets hit after a few seconds:

https://github.com/apache/mynewt-nimble/blob/32621d01b444d99dedba8fe85c44a2bb1b2460c4/nimble/transport/usb/src/ble_hci_usb.c#L207

It may very well be that things go wrong before this line as newtmgr has been stuck at 0% for a few seconds already. 

Increasing the BLE_ACL_BUF_SIZE only delays the problem, it doesn't solve it. 

Some output from btmon. The newtmgr upload appears stalled after the ACL TX at 11.608489: 

```
...
< ACL Data TX: Handle 1 flags 0x00 dlen 9                                                                                                           #173 [hci0] 11.591733
      ATT: Write Request (0x12) len 4
        Handle: 0x001a
          Data: 0100
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                              #174 [hci0] 11.602625
        Num handles: 1
        Handle: 1
        Count: 1
> ACL Data RX: Handle 1 flags 0x02 dlen 5                                                                                                           #175 [hci0] 11.606626
      ATT: Write Response (0x13) len 0
< ACL Data TX: Handle 1 flags 0x00 dlen 255                                                                                                         #176 [hci0] 11.608483
< ACL Data TX: Handle 1 flags 0x01 dlen 4                                                                                                           #177 [hci0] 11.608489
      ATT: Write Command (0x52) len 254
        Handle: 0x0019
          Data: 020000f400014201a5646461746158b13db8f3960000000020000000287903000000000000000300050000000000000000000120f98000005581000057810000000000000000000000000000000000000000000000000000000000006be100000000000000000000b1e10000f7e1000065810000658100006581000065810000658100006581000065810000658100006581000065810000658100006581000065810000658100006581000065810000658100006581000065810000658100006565696d61676500636c656e1a00037970636f666600637368615820c8cbf225c3071b62cd9570ba49d40dc7428618d5e43bd38bf28508fa76098026
< ACL Data TX: Handle 1 flags 0x00 dlen 255                                                                                                         #178 [hci0] 14.609512
< ACL Data TX: Handle 1 flags 0x01 dlen 4                                                                                                           #179 [hci0] 14.609523
      ATT: Write Command (0x52) len 254
        Handle: 0x0019
          Data: 020000f400014301a3646461746158df81000065810000658100006581000065810000658100006581000065810000658100006581000000000000000000006581000065810000658100006581000065810000658100004ff000000c4a0d4b9a42bcbf42f8040bfae70b490b4a0c4b9b1a03dd043bc858d050fbdc09480a4900f022f80948804700f03cf808480047a8140020e0c50020c8e50300d800002058140020e0c5002050fe0020a582000099840000fee7fee7fee7fee7fee7fee7fee7fee7fee700bf024bc3e900019860704700bfd8000020094a0028936806da11681844884208d3906018467047516865696d61676500636f666618b1
```

some of newtmgr output:

```
DEBU[2021-03-18 14:58:03.351] Using connection profile: name=243d type=ble connstring=peer_name=hwt_lmin-1000243D 
DEBU[2021-03-18 14:58:03.766] Connecting to peer                           
DEBU[2021-03-18 14:58:04.966] Exchanging MTU                               
DEBU[2021-03-18 14:58:04.976] Exchanged MTU; ATT MTU = 256                 
DEBU[2021-03-18 14:58:04.976] Discovering profile                          
DEBU[2021-03-18 14:58:05.171] Subscribing to NMP response characteristic   
 0 / 227696 [---------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%DEBU[2021-03-18 14:58:05.187] Encoded &{NmpBase:{hdr:{Op:2 Flags:0 Len:0 Group:1 Seq:66 Id:1}} ImageNum:0 Off:0 Len:227696 DataSha:[200 203 242 37 195 7 27 98 205 149 112 186 73 212 13 199 66 134 24 213 228 59 211 139 242 133 8 250 118 9 128 38] Upgrade:false Data:[61 184 243 150 0 0 0 0 32 0 0 0 40 121 3 0 0 0 0 0 0 0 3 0 5 0 0 0 0 0 0 0 0 0 1 32 249 128 0 0 85 129 0 0 87 129 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 107 225 0 0 0 0 0 0 0 0 0 0 177 225 0 0 247 225 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 0 0 0 0 0 0 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 101 129 0 0 79 240 0 0 12 74 13 75 154 66 188 191 66 248 4 11 250 231 11 73 11 74 12 75 155 26 3 221 4 59 200 88 208 80 251 220 9 72 10 73 0 240 34 248 9 72 128 71 0 240 60 248 8 72 0 71 168 20 0 32 224 197 0 32 200 229 3 0 216 0 0 32 88 20 0 32 224 197 0 32 80 254 0 32 165 130 0 0 153 132 0 0 254 231 254 231 254 231 254 231 254 231 254 231 254 231 254 231 254 231 0 191 2 75 195 233 0 1 152 96 112 71 0 191 216 0 0 32 9 74 0 40 147 104 6 218 17 104 24 68 136 66 8 211 144 96 24 70 112 71 81 104 201 26 129 66 1 219 24 68 246 231 79 240 255 51 244 231 216 0 0 32 79 240 128 67 1 34 195 248 120 37 112 71 8 75 27 104 219 7 0 213 1 190 191 243 79 143 6 73 6 75 202 104 2 244 224 98 19 67 203 96 191 243 79 143 0 191 253 231 240 237 0 224 0 237 0 224 4 0 250 5 2 75 24 104 0 240 1 0 112 71 0 191 240 237 0 224 79 240 128 67 1 33 89 96 211 248 24 4 11 74 16 64]} to:

...

0 B / 222.36 KiB [---------------------------------------------------------------------------------------------------------------------------------------------]   0.00%DEBU[2021-03-18 14:58:08.188] {remove-nmp-listener} [.:0] seq=66           
DEBU[2021-03-18 14:58:08.188] HandleError off 0 error NMP timeout
...
```

ncasaril, 2021-03-19:
Tweaking a buffer size in tinyusb is a workaround: 

https://github.com/hathach/tinyusb/blob/6cf110b5d0efc8556fbb7aeabef583d8c000ae4b/src/class/bth/bth_device.h#L39

This value needs to be bigger than the MTU with a margin or newtmgr will fail to upload. 
Could this 
be an issue with fragmenting the data over usb that's not taken care of in the bth_device? 

Tagging @kasjer as he's named in bth_device.c.

ncasaril, 2021-03-19:
One step closer :) 

This comparison is going in the wrong direction. The data should be sent if the actual data (om_len) is greater or equal to the expected length (len):

https://github.com/apache/mynewt-nimble/blob/32621d01b444d99dedba8fe85c44a2bb1b2460c4/nimble/transport/usb/src/ble_hci_usb.c#L214

Fixing the comparison (PR coming) removes the effect of the TU buffer. Uploads with a BLE MTU < 252 now works well. 

Still, upload fails for an MTU larger than 252, i.e. as soon as we need to fragment the messages. I've identified where in ble_ll_conn.c this seems to happen and will open a separate issue for that.

andrzej-kaczmarek, 2021-03-24:
@ncasaril could you please share your target settings for blehci?

ncasaril, 2021-03-25:
Settings listed below, note that I've added the RTT console for debugging purposes. 

```
syscfg.vals:
    CONSOLE_RTT: 1
    CONSOLE_UART: 0
    SHELL_TASK: 1
    BLE_HCI_TRANSPORT: usb
    BLE_MAX_CONNECTIONS: 32
    BLE_PUBLIC_DEV_ADDR: "(uint8_t[6]){0x20, 0x20, 0x10, 0x10, 0xBB, 0xCD}"
    USBD_BTH: 1
    USBD_PID: 0x1234
    USBD_VID: 0x5678

    BLE_EXT_ADV: 1
    BLE_EXT_ADV_MAX_SIZE: 1650
    BLE_HCI_EVT_LO_BUF_COUNT: 16
    MSYS_1_BLOCK_COUNT: 16
    MSYS_1_BLOCK_SIZE: 1024
    BLE_LL_CFG_FEAT_DATA_LEN_EXT: 1
    BLE_LL_CFG_FEAT_LE_2M_PHY: 1
    BLE_LL_CFG_FEAT_LE_CODED_PHY: 1
    BLE_LL_CFG_FEAT_LL_PRIVACY: 1
    BLE_LL_MAX_PKT_SIZE: 251
    BLE_LL_CONN_INIT_SLOTS: 4
    BLE_LL_DTM: 1
    BLE_LL_DTM_EXTENSIONS: 1
    BLE_LL_NUM_SCAN_DUP_ADVS: 32
    BLE_LL_VND_EVENT_ON_ASSERT: 1
    BLE_MAX_PERIODIC_SYNCS: 5
    BLE_MULTI_ADV_INSTANCES: 16
    BLE_PERIODIC_ADV: 1
    BLE_PERIODIC_ADV_SYNC_TRANSFER: 1
    BLE_VERSION: 51

    BLE_ACL_BUF_SIZE: 1024
    BLE_ACL_BUF_COUNT: 12
    BLE_LL_CFG_FEAT_CTRL_TO_HOST_FLOW_CONTROL: 1
```

ncasaril, 2021-03-26:
I should add that with #942 and #944 applied transfers are working up to a MTU of 512. 
Let me know if you'd prefer me to close this issue and create a new one specifically for the case of MTU > 512?

C-Innovation, 2022-01-19:
@ncasaril could you please share your source project? I have some problems to run usb transport...

Building target targets/mybtshell
Linking /home/yaroslav/dev/blehciproj/bin/targets/mybtshell/app/@apache-mynewt-nimble/apps/btshell/btshell.elf
Error: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/yaroslav/dev/blehciproj/bin/targets/mybtshell/app/@apache-mynewt-nimble/nimble/host/@apache-mynewt-nimble_nimble_host.a(ble_hs.o): in function `ble_hs_tx_data':
/home/yaroslav/dev/blehciproj/repos/apache-mynewt-nimble/nimble/host/src/ble_hs.c:711: undefined reference to `ble_hci_trans_hs_acl_tx'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/yaroslav/dev/blehciproj/bin/targets/mybtshell/app/@apache-mynewt-nimble/nimble/host/@apache-mynewt-nimble_nimble_host.a(ble_hs.o): in function `ble_hs_init':
/home/yaroslav/dev/blehciproj/repos/apache-mynewt-nimble/nimble/host/src/ble_hs.c:796: undefined reference to `ble_hci_trans_cfg_hs'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/yaroslav/dev/blehciproj/bin/targets/mybtshell/app/@apache-mynewt-nimble/nimble/host/@apache-mynewt-nimble_nimble_host.a(ble_hs_hci_cmd.o): in function `ble_hs_hci_cmd_transport':
/home/yaroslav/dev/blehciproj/repos/apache-mynewt-nimble/nimble/host/src/ble_hs_hci_cmd.c:41: undefined reference to `ble_hci_trans_hs_cmd_tx'
collect2: error: ld returned 1 exit status

sjanc, 2022-04-13:
Hi Niklas, is this til an issue?

ncasaril, 2022-04-14:
I've dropped this. I'll close the issue.

closed 2022-04-14 by ncasaril