apache/mynewt-nimble #965: [Regression/bug] Segment fault in linux porting example (Nimble 1.4.0)

verjus, 2021-04-10:
The example in apache-mynewt-nimble-1.3.0/porting/examples/linux/ works under Ubuntu 16.04.7 LTS, but the same example in 1.4.0 fails:

```
> cd apache-mynewt-nimble-1.4.0/porting/examples/linux
> make
<snip>

size nimble-linux
   text	   data	    bss	    dec	    hex	filename
 272568	   2688	  20176	 295432	  48208	nimble-linux

> sudo setcap 'cap_net_raw,cap_net_admin+eip' nimble-linux
> ./nimble-linux
Segmentation fault
> gdb ./nimble-linux
(gdb) run
Starting program: apache-mynewt-nimble-1.4.0/porting/examples/linux/nimble-linux 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff7ff5740 (LWP 25883)]

Thread 1 "nimble-linux" received signal SIGSEGV, Segmentation fault.
0x0000000000403ded in os_mempool_init_internal (mp=0x646480 <ble_hci_sock_acl_pool>, blocks=24, 
    block_size=324, membuf=0x6464e0 <ble_hci_sock_acl_buf>, name=0x4350d7 "ble_hci_sock_acl_pool", 
    flags=0 '\000') at ../../../porting/nimble/src/os_mempool.c:177
177	    STAILQ_INSERT_TAIL(&g_os_mempool_list, mp, mp_list);
```
The problem seems to originate from changes to:
apache-mynewt-nimble-1.4.0/porting/nimble/src/os_mempool.c. I tried to copy the 1.3.0 version of os_mempool.c and the program runs.

sjanc, 2021-04-11:
Hi,

should be fixed with https://github.com/apache/mynewt-nimble/pull/966

verjus, 2021-04-11:
I've tested your changes on my end and it appears to solve the issue nicely.
Thank you!

Closing this.

closed 2021-04-11 by verjus