apache/mynewt-nimble #972: Scanning on coded PHY for extended advertisements not working?

haukepetersen, 2021-04-16:
Have been trying to get a simle advertiser/scanner setup using the coded PHY working for a while now, but can't manage to get it to work. By know I have the feeling, that something in the NimBLE code base is not quite behaving as it should...

## Problem
Node A is configured as simple advertiser sending small packets with AD fields `flags` and `name` set. Node B is configured as scanner, and should in theory receive the advertising packets send by A. Both nodes are on my desk in perfect radio range of each other... When I configure them both to advertise/scan using the 1M PHY, the scanner receives all advertising packets as expected. But as soon as I switch to the coded PHY, the scanner does not see any packets anymore. Moreover, the advertiser seems to work fine, as I can see the advertisements being sent using the coded PHY with my phone (nrfconnect app @ Android).

Is there any known issue w.r.t. scanning for packets using the coded PHY?

## Setup Details
Both nodes are nrf52840-based (advertiser: `particle-xenon`, scanner: `nrf52840dk`) and are running NimBLE on RIOT using both NimBLE controller and host. NimBLE is configured as by the RIOT default configuration (see `porting/npl/riot/include/syscfg/syscfg.h`) with the following changes, same config on both nodes:
```
BLE_EXT_ADV=1
BLE_EXT_ADV_MAX_SIZE=400
BLE_LL_CFG_FEAT_LL_EXT_ADV=1
BLE_LL_EXT_ADV_AUX_PTR_CNT=5
BLE_MULTI_ADV_INSTANCES=3
BLE_LL_CFG_FEAT_LE_CODED_PHY=1
BLE_PHY_CODED_RX_IFS_EXTRA_MARGIN=1
```
 The advertiser uses the following configuration:
```c
    struct ble_gap_ext_adv_params params;
    memset(&params, 0, sizeof(params));
    params.own_addr_type = nimble_riot_own_addr_type,
    params.primary_phy = BLE_HCI_LE_PHY_CODED,
    params.secondary_phy = BLE_HCI_LE_PHY_CODED;
    params.legacy_pdu = 0;
    params.tx_power = 127;
    params.sid = 0;
    int8_t tx_power;
    int res = ble_gap_ext_adv_configure(1, &params, NULL, _on_gap_evt, NULL);
```

The scanner as follows:
```c
static struct ble_gap_ext_disc_params _disc_params = {
    .itvl = BLE_GAP_SCAN_ITVL_MS(SCAN_ITVL),
    .window = BLE_GAP_SCAN_WIN_MS(SCAN_WIN),
    .passive = 1,
};
    ...
    res = ble_gap_ext_disc(nimble_riot_own_addr_type,
                           0, 0, 0, 0, 1,
                           NULL, &_disc_params, _on_disc, NULL);
```

All used NimBLE functions return success, so at least seems not to be an obvious configuration error?! Any idea what could be going wrong here is welcome, would really much get this to work :-)

sjanc, 2021-04-19:
Hi,

I've just tested this on mynewt and all seems to work just fine.

Maybe silly question, but you are using limited discovery which means remote needs to have proper flags set in AdvData. Is this the case?

haukepetersen, 2021-04-19:
Awesome, thanks!

> Maybe silly question, but you are using limited discovery which means remote needs to have proper flags set in AdvData. Is this the case?

Yes, the proper flags are set. Also I tried it not using limited discovery without success...

But at least I know that its not a general problem and I have a direction to go looking. As this is not a general problem I close this issue again.

closed 2021-04-19 by haukepetersen

iamcalledusman, 2022-03-27:
Hi @haukepetersen!

Thank you for the nice post! I have a question can we use coded PHY for esp32c3 (esp-idf) using nimble host?
If yes can you please give me some guidelines?

Thanks in advance!