apache/mynewt-nimble #989: Multiple simultaneous connections cause reset after many PHY updates

KKopyscinski, 2021-06-15:
This was found using `blestress` app on nrf52840 DK. When test 7 (PHY updates from tx device - central) is executed it will cause silent restart during test 9 (connect and keep as many active connections as possible). Test 8 (PHY updates from rx device - peripheral) has no such influence. There is no visible sign of crash: no events captured by debugger, HCI activity just stops - this is most likely just sign of broken RTT connection on port. App is restarted just like after regular crash or reboot.

haukepetersen, 2021-11-08:
Hej, I was running in similar issues when running a node with many multiple connections. I got it to work, by increasing NimBLE's buffer space -> as I was running on RIOT, i increased `MSYS_1_BLOCK_COUNT` as much as my RAM would allow me to. But I guess this could be easily done somehow on Mynewt as well, right? With this I was successfully running node with 14 simultaneous connections...

However, I could never figure out, where and why NimBLE crashes once it runs out of buffers, nor if it is the controller or the host part causing the trouble. But I think on the long term it would be good to locate this issue and fix it, so maybe someone else has some more luck in finding it?!

andrzej-kaczmarek, 2021-11-08:
this will fix it https://github.com/apache/mynewt-nimble/pull/1081

rx side was reset due to watchdog timeout caused by broken callouts queue in test 7. there was callout enqueued which was then wiped out and it expired somewhere during test 9, and it could not be removed from queue so it "expired" over and over again causing watchdog reset.

closed 2023-09-26 by sjanc