apache/mynewt-nimble #1020: Update Nimble to PTS 8.1

rymanluk, 2021-09-09:
- [ ] Make sure GAP test cases passes and contains all new ones
- [ ]  Same with GATT
- [ ] Same with SM
- [ ] Same with L2CAP
- [ ] Same with Mesh

KKopyscinski, 2021-09-28:
Update of PTS workspace in autopts:
https://github.com/intel/auto-pts/pull/592

First batch of fixed tests in autopts:
https://github.com/intel/auto-pts/pull/590

NimBLE fixes:
https://github.com/apache/mynewt-nimble/pull/1024
https://github.com/apache/mynewt-nimble/pull/1025
https://github.com/apache/mynewt-nimble/pull/1032

KKopyscinski, 2021-09-28:
FYI @jrotkiewicz

KKopyscinski, 2021-11-19:
Pending fixes:
- Host:
   - https://github.com/apache/mynewt-nimble/pull/1099
   - https://github.com/apache/mynewt-nimble/pull/1098
   - https://github.com/apache/mynewt-nimble/pull/1003
- bttester
   - https://github.com/apache/mynewt-nimble/pull/1096
   - https://github.com/apache/mynewt-nimble/pull/1094
   - https://github.com/apache/mynewt-nimble/pull/1091
- autopts:
   - https://github.com/intel/auto-pts/pull/590

KKopyscinski, 2021-11-29:
With submitted PR test results are as follows:

<details>
    <summary>Results</summary>

| Group | Test | Verdict |
|---------|----------|----------|
GAP | GAP/BROB/BCST/BV-01-C | PASS
GAP | GAP/BROB/BCST/BV-02-C | PASS
GAP | GAP/BROB/OBSV/BV-01-C | PASS
GAP | GAP/BROB/OBSV/BV-02-C | PASS
GAP | GAP/BROB/OBSV/BV-05-C | PASS
GAP | GAP/BROB/OBSV/BV-06-C | PASS
GAP | GAP/DISC/NONM/BV-01-C | PASS
GAP | GAP/DISC/NONM/BV-02-C | PASS
GAP | GAP/DISC/LIMM/BV-03-C | PASS
GAP | GAP/DISC/LIMM/BV-04-C | PASS
GAP | GAP/DISC/GENM/BV-04-C | PASS
GAP | GAP/DISC/LIMP/BV-01-C | PASS
GAP | GAP/DISC/LIMP/BV-02-C | PASS
GAP | GAP/DISC/LIMP/BV-03-C | PASS
GAP | GAP/DISC/LIMP/BV-04-C | PASS
GAP | GAP/DISC/LIMP/BV-05-C | PASS
GAP | GAP/DISC/GENP/BV-01-C | PASS
GAP | GAP/DISC/GENP/BV-02-C | PASS
GAP | GAP/DISC/GENP/BV-03-C | PASS
GAP | GAP/DISC/GENP/BV-04-C | PASS
GAP | GAP/DISC/GENP/BV-05-C | PASS
GAP | GAP/DISC/RPA/BV-01-C | PASS
GAP | GAP/IDLE/NAMP/BV-01-C | PASS
GAP | GAP/IDLE/NAMP/BV-02-C | PASS
GAP | GAP/CONN/NCON/BV-01-C | PASS
GAP | GAP/CONN/NCON/BV-02-C | PASS
GAP | GAP/CONN/NCON/BV-03-C | PASS
GAP | GAP/CONN/DCON/BV-01-C | PASS
GAP | GAP/CONN/UCON/BV-01-C | PASS
GAP | GAP/CONN/UCON/BV-02-C | PASS
GAP | GAP/CONN/UCON/BV-03-C | PASS
GAP | GAP/CONN/ACEP/BV-01-C | INCONC
GAP | GAP/CONN/ACEP/BV-03-C | INCONC
GAP | GAP/CONN/GCEP/BV-01-C | INCONC
GAP | GAP/CONN/GCEP/BV-02-C | PASS
GAP | GAP/CONN/SCEP/BV-01-C | INCONC
GAP | GAP/CONN/SCEP/BV-03-C | INCONC
GAP | GAP/CONN/DCEP/BV-01-C | INCONC
GAP | GAP/CONN/DCEP/BV-03-C | PASS
GAP | GAP/CONN/CPUP/BV-08-C | PASS
GAP | GAP/CONN/CPUP/BV-10-C | PASS
GAP | GAP/CONN/TERM/BV-01-C | PASS
GAP | GAP/CONN/PRDA/BV-01-C | PASS
GAP | GAP/CONN/PRDA/BV-02-C | PTS TIMEOUT
GAP | GAP/BOND/NBON/BV-01-C | PASS
GAP | GAP/BOND/NBON/BV-02-C | PASS
GAP | GAP/BOND/NBON/BV-03-C | PASS
GAP | GAP/BOND/BON/BV-01-C | PASS
GAP | GAP/BOND/BON/BV-02-C | PASS
GAP | GAP/BOND/BON/BV-03-C | PASS
GAP | GAP/BOND/BON/BV-04-C | PASS
GAP | GAP/SEC/SEM/BV-45-C | PASS
GAP | GAP/SEC/AUT/BV-11-C | PASS
GAP | GAP/SEC/AUT/BV-12-C | PASS
GAP | GAP/SEC/AUT/BV-13-C | PASS
GAP | GAP/SEC/AUT/BV-14-C | PASS
GAP | GAP/SEC/AUT/BV-17-C | FAIL
GAP | GAP/SEC/AUT/BV-18-C | PASS
GAP | GAP/SEC/AUT/BV-19-C | INCONC
GAP | GAP/SEC/AUT/BV-20-C | PASS
GAP | GAP/SEC/AUT/BV-21-C | PASS
GAP | GAP/SEC/AUT/BV-22-C | PASS
GAP | GAP/SEC/AUT/BV-23-C | PASS
[... 562 lines cut ...]
- Passing with use of LE dongle:
  - GAP/CONN/ACEP/BV-03-C
  - GAP/CONN/SCEP/BV-03-C
  - GAP/CONN/SCEP/BV-03-C
  - GAP/CONN/SCEP/BV-03-C
  - GAP/CONN/ACEP/BV-04-C
  - GAP/CONN/GCEP/BV-05-C
  - GAP/CONN/GCEP/BV-06-C
  - GAP/CONN/DCEP/BV-05-C
  - GAP/CONN/DCEP/BV-05-C
- Errata:
  - GAP/CONN/ACEP/BV-01-C		CASE0072061
  - GAP/CONN/GCEP/BV-01-C		CASE0072061
  - GAP/CONN/SCEP/BV-01-C		CASE0072061
  - GAP/CONN/DCEP/BV-01-C		CASE0072061
  - GAP/SEC/AUT/BV-19-C		        CASE0072154
  - L2CAP/LE/CFC/BV-29-C		        CASE0071766
  - L2CAP/ECFC/BV-24-C		        TSE14759 (TCW1333)
  - L2CAP/ECFC/BV-38-C		        TSE17458
  - GAP/SEC/AUT/BV-25-C		        CASE0072001
  - GAP/SEC/AUT/BV-27-C		        CASE0072001
- Passing with submitted fixes (from previous comment)
  - GAP/CONN/SCEP/BV-01-C
  - GAP/CONN/PRDA/BV-02-C
  - GAP/SEC/AUT/BV-17-C

sjanc, 2021-11-29:
I think we should just assume that LE dongles are used for LE testing from no on (ie don't care if test is failing on old dual mode dongle)

closed 2022-11-03 by sjanc