apache/mynewt-nimble #1023: linux_blemesh failed

donny681, 2021-09-14:
**Problem**
I don't know the reason why fail to create free memslab list?I test the linux_blemesh demo.The ble device is ESP32 hci device(support BLE4.2).It success to run the nimble-linux demo.
`yates@yates-virtual-machine:~/software/mynewt-nimble/porting/examples/linux_blemesh$ sudo ./nimble-linux-blemesh 
GAP procedure initiated: stop advertising.

Bluetooth initialized

bt_mesh_trans_init: Failed to create free memslab list
Mesh initialized
`

**Step**
1.Btattach the ESP32  hci device to the linux 
`sudo btattach -B /dev/ttyUSB1 -P h4 -S 921600
`
2.verify the system can recognize the esp32 device and get the ble mac
`yates@yates-virtual-machine:~/software/mynewt-nimble/porting/examples/linux_blemesh$ hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 30:AE:A4:1A:F7:2E  ACL MTU: 1021:9  SCO MTU: 255:4

	UP RUNNING PSCAN ISCAN 

	RX bytes:7926 acl:0 sco:0 events:414 errors:0

	TX bytes:23009 acl:16 sco:0 commands:485 errors:0

	Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87

	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3 

	Link policy: RSWITCH SNIFF 

	Link mode: SLAVE ACCEPT 

	Name: 'yates-virtual-machine'

	Class: 0x1c0000

	Service Classes: Rendering, Capturing, Object Transfer

	Device Class: Miscellaneous, 

	HCI Version: 4.2 (0x8)  Revision: 0x30e

	LMP Version: 4.2 (0x8)  Subversion: 0x30e

	Manufacturer: RivieraWaves S.A.S (96)`

donny681, 2021-09-14:
I check hcidump for hci0 ,It show bottom information.

`HCI sniffer - Bluetooth packet analyzer ver 5.53
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
> HCI Event: Command Complete (0x0e) plen 4
    Reset (0x03|0x0003) ncmd 5
    status 0x00
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 5
    status 0x00
    HCI Version: 4.2 (0x8) HCI Revision: 0x30e
    LMP Version: 4.2 (0x8) LMP Subversion: 0x30e
    Manufacturer: RivieraWaves S.A.S (96)
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Supported Features (0x04|0x0003) ncmd 5
    status 0x00
    Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 5
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask Page 2 (0x03|0x0063) ncmd 5
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 5
    status 0x00
> HCI Event: Command Complete (0x0e) plen 7
    LE Read Buffer Size (0x08|0x0002) ncmd 5
    status 0x00 pktlen 0x00fb maxpkt 0x0a
> HCI Event: Command Complete (0x0e) plen 12
    LE Read Local Supported Features (0x08|0x0003) ncmd 5
    status 0x00 features 0xff00000000000000 (Link Layer supports LE Encryption)
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 5
    status 0x00 bdaddr 30:AE:A4:1A:F7:2E
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x002d) ncmd 5
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x0029) ncmd 5
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x002d) ncmd 5
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 5
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x0027) ncmd 5
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x004e) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
> HCI Event: Command Complete (0x0e) plen 12
    LE Rand (0x08|0x0018) ncmd 5
`

sjanc, 2021-10-04:
RFC fix:  https://github.com/apache/mynewt-nimble/pull/1044

closed 2021-11-02 by sjanc