apache/mynewt-nimble #1046: How to build mynewt-nimble for linux platform by newt commands ?

gavin-yy, 2021-10-05:
hello,
    Can i build mynewt-nimble for linux platform by newt commands ? When i am trying to create a target (for blehr), If i set the target.bsp field as 'bsp/native' in the target.yml, it requires a ble driver ?
    The demo on linux can only be built by makefile under apache-mynewt-nimble/porting/examples/linux currently.

sjanc, 2021-10-06:
Hello,

This was answered on ML, wasn't it?

https://lists.apache.org/thread.html/ra5c42ee587f89c936775c58c0bfc4f94900820a33664dd93058a9d87%40%3Cdev.mynewt.apache.org%3E

closed 2021-10-06 by sjanc

gavin-yy, 2021-10-08:
tried method1: 
```shell
newt build -l DEBUG  @apache-mynewt-nimble/porting/targets/linux
repos/apache-mynewt-core/kernel/os/include/os/os_time.h:65:10: fatal error: os/os_arch.h: No such file or directory
~~ missing many files after fix some marco defines.
```
method2:
```
change apache-mynewt-nimble/porting/targets/linux/target.yml to:
target.app: "porting/targets/dummy_app"
target.bsp: "@apache-mynewt-core/hw/bsp/native"
target.build_profile: "debug"

newt build -l DEBUG  @apache-mynewt-nimble/porting/targets/linux
dummy app didn't building porting/npl/linux files, the ble feature is not really available.
```
so i think the linux porting missed mewt package defines. It should be added.  and a runable demo should be created base on porting/example/linux if we want to build it by newt commands (currently it can be built by makefile and works fine).

sjanc, 2021-10-08:
Yes, although you may tune configuration by editing include/syscfg/syscfg.h