apache/mynewt-nimble #1049: bug: legacy scan always returns type BLE_HCI_ADV_TYPE_ADV_DIRECT_IND_LD

haukepetersen, 2021-10-05:
After rebasing to the current NimBLE master I was wondering, that my BLE node was only able to scan for low-duty cycle DIRECT_IND messages, and no other messages were found by the scanning node anymore... After a short `git bisect` session it turned out, that eed4389851fcf96922a2e429376808f2dfddc912 does seem to break something.

Building against e084ef70efdb62ddff25a5f97b024561bf4d8de6, the output of my RIOT-NimBLE Scanner looks something like:
```
2021-10-05 20:00:12,019 # ble scan 2000                                                                                
2021-10-05 20:00:12,021 # scanning (for 2000ms) ...                                                                    
2021-10-05 20:00:14,022 # done                                                                                         
2021-10-05 20:00:14,031 # [ 0] 7F:A7:8E:A1:95:5E (random)  [IND] "undefined", adv_msg_cnt: 4, adv_int: 203904us, last_rssi: -83
2021-10-05 20:00:14,040 # [ 1] A7:74:A9:EA:28:0A (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 1, adv_int: 0us, last_rssi: -78                                                                                                            
2021-10-05 20:00:14,049 # [ 2] F0:E5:70:D0:A4:4C (random)  [IND] "undefined", adv_msg_cnt: 33, adv_int: 57061us, last_rssi: -64                                                                                                               
2021-10-05 20:00:14,058 # [ 3] DB:E4:F4:C2:21:DA (random)  [IND] "undefined", adv_msg_cnt: 30, adv_int: 62887us, last_rssi: -68
2021-10-05 20:00:14,067 # [ 4] D0:D0:03:70:8A:A9 (public)  [IND] "undefined", adv_msg_cnt: 13, adv_int: 130771us, last_rssi: -74
2021-10-05 20:00:14,077 # [ 5] 28:39:5E:32:5A:30 (public)  [NONCONN_IND] "undefined", adv_msg_cnt: 5, adv_int: 309750us, last_rssi: -80                                                                                                       
2021-10-05 20:00:14,086 # [ 6] 0C:61:CF:3F:D2:B7 (public)  [IND] "undefined", adv_msg_cnt: 4, adv_int: 288016us, last_rssi: -82                                                                                                               
2021-10-05 20:00:14,096 # [ 7] CC:AC:6B:4A:AE:DF (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 2, adv_int: 502299us, last_rssi: -67
2021-10-05 20:00:14,105 # [ 8] 46:8B:FE:58:3A:2E (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 2, adv_int: 504828us, last_rssi: -72
2021-10-05 20:00:14,115 # [ 9] FC:76:A8:C4:1D:CF (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 2, adv_int: 497806us, last_rssi: -73
2021-10-05 20:00:14,125 # [10] A5:C9:39:E0:3B:B6 (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 2, adv_int: 500964us, last_rssi: -66
```

But building the same application against eed4389851fcf96922a2e429376808f2dfddc912 and running it yields:
```
2021-10-05 20:01:17,771 # ble scan 2000
2021-10-05 20:01:17,774 # scanning (for 2000ms) ...
2021-10-05 20:01:19,775 # done
2021-10-05 20:01:19,784 # [ 0] 0C:61:CF:3F:D2:B7 (public)  [DIRECT_IND_LD] "undefined", adv_msg_cnt: 1, adv_int: 0us, last_rssi: -81
2021-10-05 20:01:19,794 # [ 1] F0:E5:70:D0:A4:4C (random)  [DIRECT_IND_LD] "undefined", adv_msg_cnt: 14, adv_int: 134605us, last_rssi: -55
2021-10-05 20:01:19,803 # [ 2] 7A:FC:62:2E:E4:AE (random)  [DIRECT_IND_LD] "undefined", adv_msg_cnt: 6, adv_int: 226151us, last_rssi: -71
2021-10-05 20:01:19,813 # [ 3] DB:E4:F4:C2:21:DA (random)  [DIRECT_IND_LD] "undefined", adv_msg_cnt: 15, adv_int: 120060us, last_rssi: -65
2021-10-05 20:01:19,825 # [ 4] D0:D0:03:70:8A:A9 (public)  [DIRECT_IND_LD] "[TV] Samsung 7 Series (55)", adv_msg_cnt: 3, adv_int: 569803us, last_rssi: -75
````
-> now the `event_type` field of `struct ble_gap_disc_desc` is always set to `BLE_HCI_ADV_TYPE_ADV_DIRECT_IND_LD`.

andrzej-kaczmarek, 2021-10-05:
That is strange, it works just fine for me with and without ext adv enabled. Also there were no regressions on qualification test cases after that change. What scan settings do you use by default?

haukepetersen, 2021-10-06:
Strange indeed...

In this particular case I am using the following scan parameters:
```c
struct ble_gap_disc_params p = { 0 };
p.itvl = BLE_GAP_SCAN_ITVL_MS(100);
p.window = BLE_GAP_SCAN_WIN_MS(10);
```

andrzej-kaczmarek, 2021-10-06:
Do you have any special host/controller settings for that app?

BTW, `event_type` uses `BLE_HCI_ADV_RPT_EVTYPE_xxx` defines - `BLE_HCI_ADV_TYPE_ADV_xxx` are used by advertiser to configure advertising type. So since `BLE_HCI_ADV_TYPE_ADV_DIRECT_IND_LD=4` I assume what you are seeing is `BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP=4`

haukepetersen, 2021-10-06:
> BTW, event_type uses BLE_HCI_ADV_RPT_EVTYPE_xxx defines - BLE_HCI_ADV_TYPE_ADV_xxx are used by advertiser to configure advertising type. So since BLE_HCI_ADV_TYPE_ADV_DIRECT_IND_LD=4 I assume what you are seeing is BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP=4

Yes, you assume correct. Seems like this wrong define mapping was in the RIOT code base for a while now, fixed it while we are at it...

But back to the issue at hand: I am not using any particular host/controller settings regarding GAP discovery. Extended advertisements are disabled, else the default configuration for NImBLE in RIOT is used (-> `porting/targets/riot/syscfg.yml`).

Just to make sure, I just ran another round of tests: I compiled the `examples/nimble_scanner` example in RIOT (branch https://github.com/RIOT-OS/RIOT/pull/16955 to include the fixses as discussed above) against different NimBLE commits and ran the example on a `nrf52840dk` board. The results are the same than yesterday evening:

building agains eed4389851fcf96922a2e429376808f2dfddc912 and newer:
```
2021-10-06 17:10:22,696 # scan 3000                        
2021-10-06 17:10:25,699 # Scanning for 3000 ms now ... done 
2021-10-06 17:10:25,699 #                                  
2021-10-06 17:10:25,700 # Results:
2021-10-06 17:10:25,709 # [ 0] 5F:85:20:0E:ED:F3 (random)  [SCAN_RSP] "undefined", adv_msg_cnt: 1, adv_int: 0us, last_rssi: -83
2021-10-06 17:10:25,720 # [ 1] D0:D0:03:70:8A:A9 (public)  [SCAN_RSP] "[TV] Samsung 7 Series (55)", adv_msg_cnt: 5, adv_int: 552510us, last_rssi: -76
2021-10-06 17:10:25,729 # [ 2] F0:E5:70:D0:A4:4C (random)  [SCAN_RSP] "undefined", adv_msg_cnt: 21, adv_int: 132300us, last_rssi: -71
```

building against e084ef70efdb62ddff25a5f97b024561bf4d8de6 and older:
```
2021-10-06 17:10:03,458 # scan 3000                                                                                    
2021-10-06 17:10:06,461 # Scanning for 3000 ms now ... done 
2021-10-06 17:10:06,461 #                                                                                              
2021-10-06 17:10:06,475 # Results:     
2021-10-06 17:10:06,476 # [ 0] 5A:30:67:76:D9:DD (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 1, adv_int: 0us, last_rssi: -85
2021-10-06 17:10:06,481 # [ 1] 28:39:5E:32:5A:30 (public)  [NONCONN_IND] "undefined", adv_msg_cnt: 14, adv_int: 201365us, last_rssi: -81
2021-10-06 17:10:06,490 # [ 2] F6:EF:C4:B9:23:11 (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 3, adv_int: 318824us, last_rssi: -73
2021-10-06 17:10:06,499 # [ 3] D0:D0:03:70:8A:A9 (public)  [IND] "undefined", adv_msg_cnt: 8, adv_int: 342623us, last_rssi: -80
2021-10-06 17:10:06,509 # [ 4] 1F:3D:04:B2:27:BB (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 5, adv_int: 542639us, last_rssi: -77
2021-10-06 17:10:06,518 # [ 5] F0:E5:70:D0:A4:4C (random)  [IND] "undefined", adv_msg_cnt: 45, adv_int: 62745us, last_rssi: -61
2021-10-06 17:10:06,528 # [ 6] CC:AC:6B:4A:AE:DF (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 3, adv_int: 669017us, last_rssi: -56
2021-10-06 17:10:06,538 # [ 7] A5:C9:39:E0:3B:B6 (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 3, adv_int: 669667us, last_rssi: -60
2021-10-06 17:10:06,547 # [ 8] 4D:C7:E4:E9:25:BE (random)  [IND] "undefined", adv_msg_cnt: 3, adv_int: 723306us, last_rssi: -83
2021-10-06 17:10:06,556 # [ 9] A7:74:A9:EA:28:0A (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 2, adv_int: 498911us, last_rssi: -82
2021-10-06 17:10:06,566 # [10] 46:8B:FE:58:3A:2E (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 4, adv_int: 251988us, last_rssi: -71
2021-10-06 17:10:06,575 # [11] FC:76:A8:C4:1D:CF (random)  [NONCONN_IND] "undefined", adv_msg_cnt: 1, adv_int: 0us, last_rssi: -70
```

Next to the supposively wrongly reported event type, it is noticable that a lot of discovery event are not event reported as it seems...

The code triggering the scanner from RIOT is quite straight forward:
```c
    struct ble_gap_disc_params _scan_params = {
        .itvl = BLE_GAP_LIM_DISC_SCAN_INT,
        .window = BLE_GAP_LIM_DISC_SCAN_WINDOW,
        .filter_policy = 0,                         /* don't use */
        .limited = 0,                               /* no limited discovery */
        .passive = 0,                               /* no passive scanning */
        . filter_duplicates = 0,                    /* no duplicate filtering */
    };

...

    if (ble_gap_disc_active() == 0) {
        int res = ble_gap_disc(nimble_riot_own_addr_type, _scan_duration,
                               &_scan_params, _on_scan_evt, NULL);
        if (res != 0) {
            DEBUG("[scanner] err: start failed (%i)\n", res);
            return -ECANCELED;
        }
    }
```
The only thing notable here is that we use quite aggressive scan parameters in this application, but this should not influence the reported event types, right?!

andrzej-kaczmarek, 2021-10-06:
@haukepetersen we reproduced this locally and here's the fix: https://github.com/apache/mynewt-nimble/pull/1051 :)