apache/mynewt-nimble #1076: ll: assertion fail in ble_ll_scan_aux.c

haukepetersen, 2021-11-05:
We currently run some RIOT nodes that use extended advertisements to communicate with each other. In particular, the advertisements are either `directed` or `undirected`, and all of them are `non-connectable` and `non-scannable`. In our configuration, the nodes are scanning with a 100% duty cycle (scan itvl := scan win) and are advertising data in short bursts every second or so.

Now when running a couple of nodes, they crash after random amounts of time, all showing the same assertion failure, e.g.:
```
1636113458.598309;nrf52dk-10;build/pkg/nimble/nimble/controller/src/ble_ll_scan_aux.c:1718 => 0x779                    
1636113458.600407;nrf52dk-10;*** RIOT kernel panic:                                                                    
1636113458.602457;nrf52dk-10;FAILED ASSERTION.  
```

I am pretty sure we were able to run a similar setup without any trouble some months ago, so could this be related to the latest restructuring of code in the controller?

I have tried to go through the controller code, but I can't find an obvious reason for `aux_data_current` being set `NULL` before `ble_ll_scan_aux_halt()` is called. Could it be a concurrency issue? If I see it correctly, `ble_ll_scan_aux_halt` is only used from interrupt context. And also all functions that change the value of `aux_data_current` are executed in the same context, right? 

Any idea what could be causing our nodes to crash?

For completeness, we use the RIOT default configuration (as in `porting/npl/riot/include/syscfg/syscfg.h`), with the following changes:
```
MYNEWT_VAL_MSYS_1_BLOCK_SIZE=294
MYNEWT_VAL_MSYS_1_BLOCK_COUNT=32
MYNEWT_VAL_BLE_MULTI_ADV_INSTANCES=2
MYNEWT_VAL_BLE_EXT_ADV=1
MYNEWT_VAL_BLE_HCI_EVT_HI_BUF_COUNT=12
MYNEWT_VAL_BLE_LL_EXT_ADV_AUX_PTR_CNT=10
MYNEWT_VAL_BLE_LL_CFG_FEAT_LL_EXT_ADV=1
MYNEWT_VAL_BLE_HCI_EVT_BUF_SIZE=257
MYNEWT_VAL_BLE_EXT_ADV_MAX_SIZE=1650
```

andrzej-kaczmarek, 2021-11-05:
@haukepetersen yeah, this is some regression after recent refactoring done in aux scanner. this should fix it [`8bcd2db` (#1077)](https://github.com/apache/mynewt-nimble/pull/1077/commits/8bcd2db1105352a99fe575e2ae4490178970d8ef).

haukepetersen, 2021-11-08:
Nice, thanks! Will run some tests today and report back one I have the results.

haukepetersen, 2021-11-09:
fixed by #1077

closed 2021-11-09 by haukepetersen