apache/mynewt-nimble #1122: OOB write in ble_ll_read_tx_power?

Avamander, 2021-12-05:
```
In function 'ble_ll_read_tx_power',
    inlined from 'ble_ll_hci_le_cmd_proc' at ./Pinetime/src/libs/mynewt-nimble/nimble/controller/src/ble_ll_hci.c:1154:14,
    inlined from 'ble_ll_hci_cmd_proc' at ./Pinetime/src/libs/mynewt-nimble/nimble/controller/src/ble_ll_hci.c:1585:14: ./Pinetime/src/libs/mynewt-nimble/nimble/controller/src/ble_ll_hci.c:773:23: warning: writing 2 bytes into a region of size 0 [-Wstringop-overflow=]
  773 |     rsp->min_tx_power = ble_phy_txpower_round(-127);
```

I was doing LTO on my NimBLE build and encountered this error. 

Unfortunately I can not say if it is entirely correct, is it?

sjanc, 2021-12-06:
Hi,

we are re-using command buffer for response so maybe this is confusing LTO...  from a quick look at the code I don't see anything unusual about ble_ll_read_tx_power().  Is this the only function that is causing issues?

Avamander, 2021-12-06:
Yep, that's the only warning I saw.

Avamander, 2021-12-06:
If the write is fine then I guess `gcc-arm-none-eabi-10.3-2021.10` is having hard time reasoning about it, and maybe some alternative syntax or a cast would fix it?

sjanc, 2021-12-08:
I quickly check with lto enabled and I don't get this particular error on arm-none-eabi-gcc (Fedora 10.2.0-4.fc34) 10.2.0
(I get other errors though, mostly due to signatures mismatch and some linking stuff)

Avamander, 2021-12-19:
I will consider it as a temporary fluke caused by the specific usage of NimBLE. A bunch of commits ago I could reliably get the warning shown, but no longer.

closed 2021-12-19 by Avamander