apache/mynewt-nimble #1135: Unnecessary overlap caused by the 32768Hz crystal

sada45, 2021-12-19:
Hi!
I'm reading the source code of the NimBLE link layer. I found there are unnecessary overlaps even if **all connection has a same connection interval and NimBLE trys to avoid the overlap when allocate the first anchor point.** 

1 CPU tick represent 30.517 usec.  Therefore, the connection state machine use `anchor_point` and `anchor_point_usec` to present the accurate anchor point position. As for the link layer, only the cpu ticks are used to schedule the schedule-items.

When the NimBLE trys to allocate the first anchor point for a new connection, the link layer trys to place the new anchor point **right after** the existing connection events. The corresponding code is in `ble_ll_sched.c` 
```C
sch->start_time = entry->end_time;
```
There is no overlap at the beginning, and the offset between anchor points is **164** ticks, exactly same as the duration of the schedule-item. However, as time goes on, the offset can be periodically decreased to **163** ticks because one of these two connection state machines trys to fix the  inaccuracy of the 32768 Hz crystal. Therefore, threre comes up a unnecessary overlap between these two connections. 

To my understanding, the simplest way to fix this is change the code above to:
```C
sch->start_time = entry->end_time + 1;
```

I'm using the NimBLE intergrated in the RIOT OS and the solution works fine! 
Althogh the NimBLE used in the RIOT OS is not the latest version, the latest NimBLE still uses the same strategy.

sjanc, 2021-12-30:
Hi,

Proposed fix looks reasonable to me, could you send PR with this?

sada45, 2021-12-31:
Hi sjanc,
The PR is sent
Happy new year!

closed 2022-01-19 by sjanc