apache/mynewt-nimble #1198: Missing interface to specify channel select algorithm

kiowadriver, 2022-03-04:
It is unclear from the documentation how to specify the channel select algorithm, specifically how to specify algo #2 for advertisement.

Can an example be included in the example?

sjanc, 2022-03-07:
Hi,

You cannot specify which algorithm (CSA) to use, on primary channels there is no "CSA", on secondary channels for extended advertising we use it as convenient feature(spec doesn't specify how to select channel), on periodic advertising it is mandatory to use CSA2

For connections, if enabled CSA2 is used if both sides support it.

kiowadriver, 2022-03-07:
If the peripheral supports it and is advertising with CSA2, yet the central
does not support it, will a connection still be possible?  Will the
peripheral default back to CSA1?

sjanc, 2022-03-08:
yes, CSA2 is used on connection only if both sides support it

closed 2022-03-17 by sjanc