apache/mynewt-nimble #1207: NRF52: Use with MCU_LFCLK_SOURCE: LFRC

StarGate01, 2022-03-16:
Hi, I want to use Nimble on a NRF52832 board which does not contain an external low frequency crystal (LFXO) - specifically, I want to build InfiniTime for a P8 smartwatch variant. To generate the low frequency clock, a signal can either be synthesized using the internal high frequency clock (LFSYNT), or it can be derived from an integrated RC clock (LFRC): https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fclock.html .

Using LFSYNT works great, however I would like to use the RC clock (LFRC) due to the high energy usage of the high frequency clock. Unfortunately, Nimble does not function correctly in this mode. Although it does advertise itself via BLE, no connection can be established.

The documentation at https://mynewt.apache.org/latest/network/ble_setup/ble_lp_clock.html does not mention the RC clock anywhere. Is this an unsupported mode? How could I enable this mode? Is these any example code which is known to be working in this configuration? Thanks for any help.

andrzej-kaczmarek, 2022-03-16:
LFRC needs to be calibrated to improve accuracy to 500ppm before it can be used as LP clock source for NimBLE. You need to set `BLE_LL_SCA: 500` to let NimBLE know accuracy of sleep clock (i.e. 500ppm) and then perform LFRC calibration (I don't think Mynewt does it automatically).

StarGate01, 2022-03-17:
Thank you for the help. I configured the ppm parameter to be 500, and implemented a simple calibration loop:

```
void calibrate_loop(nrf_drv_clock_evt_type_t event) {
  // 16 * 0.25s = 4s calibration cycle
  nrf_drv_clock_calibration_start(16, calibrate_loop);
}
...
nrf_drv_clock_lfclk_request(NULL);
nrfx_clock_lfclk_start();
#if(CLOCK_CONFIG_LF_SRC == NRF_CLOCK_LFCLK_RC)
    nrf_drv_clock_calibration_start(0, calibrate_loop);
#endif
```
I have confirmed the LFRC clock to be running by inspecting the registers, and the Nimble BLE link layer seems to be receiving IRQ events from the LPRC associated RTC0 (I think TIMER5 is used) as well.

However, connecting via Bluetooth still fails. Are there any other options I have to configure? Is it possible to somehow validate or debug the timing requirements?

fanoush, 2022-03-17:
[here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Fhardware_driver_clock.html&cp=4_0_0_2_0_0_3&anchor=hardware_driver_clock_calibration)  is some piece of code to do the calibration that looks a bit different can it be that the calibration loop you have there is somehow stalled and doesn't work correctly?

StarGate01, 2022-03-18:
Thanks for the link, I adjusted the calibration loop to this:

```c++
void calibrate_loop(nrf_drv_clock_evt_type_t event) {
  // 10 * 0.25s = 2.5s calibration cycle
  if (event == NRF_DRV_CLOCK_EVT_CAL_DONE) {
    NRF_LOG_INFO("Calibration done, scheduling next one");
    nrf_drv_clock_calibration_start(10, calibrate_loop);
  }
}
```

I enabled debug output for Nimble and the NRF SDK. It seems that the initialization and calibration loop does its job, at least as far as the NRF SDK is concerned.

```
<info> CLOCK: Function: nrfx_clock_init, error code: NRF_SUCCESS.
<info> CLOCK: Module enabled.
<info> clock: Function: nrf_drv_clock_init, error code: NRF_SUCCESS.
<warning> clock: Function: nrf_drv_clock_calibration_start, error code: NRF_SUCCESS.
<warning> CLOCK: Function: nrfx_clock_calibration_start, error code: NRF_SUCCESS.
<info> app: Calibration done, scheduling next one
```

The following log prints every 2.5s (Lines might be out of order due to threading or RTT buffer latency)
```
<warning> CLOCK: Function: nrfx_clock_calibration_start, error code: NRF_SUCCESS.
<info> app: Calibration done, scheduling next one
<warning> clock: Function: nrf_drv_clock_calibration_start, error code: NRF_SUCCESS.
```

The GAP advertisment call is logged as follows:
```
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
```

NRF connect on my phone does only sporadically receive these advertisements.

Once a peer tries to connect, the following is logged:
```
<info> app: Advertising event : BLE_GAP_EVENT_CONNECT
<info> app: connection established; status=0 
<info> app: [SPIMASTER] Wakeup
<info> app: [TWIMASTER] Wakeup
<info> app: [SpiNorFlash] ID on Wakeup: 21
<info> app: [SpiNorFlash] Wakeup
<info> app: Calibration done, scheduling next one
<warning> clock: Function: nrf_drv_clock_calibration_start, error code: NRF_SUCCESS.
<info> app: [LCD] Wakeup
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
<info> app: Advertising event : BLE_GAP_EVENT_DISCONNECT
<info> app: disconnect; reason=574
GATT procedure initiated: discover service by uuid; uuid=0x1805
<info> app: [Discovery] Starting discovery
<info> app: [Discovery] Discover next service
<info> app: [CTS] Starting discovery
```
According to https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html, a return code of 574 means "Connection Failed to be Established".

Could the calibration cycle interfere with the Bluetooth functionality?

InfiniTime uses FreeRTOS instead of mynewt, but still uses nimble as a library. Could that cause timing issues?

For reference, I attached two more logs with timestamps. One with LFSYNT and one with LFRC. Using LFSYNT, NRF connect receives an advertisement reliably every ~60 ms, and everything else Bluetooth works as well.

<details>
<summary>LFSYNT</summary>

```
[2022-03-17T23:03:13.800Z, +020441ms] <info> CLOCK: Function: nrfx_clock_init, error code: NRF_SUCCESS.
[2022-03-17T23:03:13.800Z, +020441ms] <info> CLOCK: Module enabled.
[... 389 lines cut ...]

</details>

Using a debugger, I caught the call to be returned from at `nimble/controller/src/ble_ll_conn.c:ble_ll_conn_event_end`
```c++
if (connsm->conn_state == BLE_LL_CONN_STATE_CREATED) {
    tmo = (uint32_t)connsm->conn_itvl * BLE_LL_CONN_ITVL_USECS * 6UL;
    ble_err = BLE_ERR_CONN_ESTABLISHMENT;
} else {
    tmo = connsm->supervision_tmo * BLE_HCI_CONN_SPVN_TMO_UNITS * 1000UL;
    ble_err = BLE_ERR_CONN_SPVN_TMO;
}
/* XXX: Convert to ticks to usecs calculation instead??? */
tmo = os_cputime_usecs_to_ticks(tmo);
if ((int32_t)(connsm->anchor_point - connsm->last_rxd_pdu_cputime) >= tmo) {
    ble_ll_conn_end(connsm, ble_err);
    return;
}
```

with `ble_err == BLE_ERR_CONN_ESTABLISHMENT`.

Since this code differs from the upstream nimble, (according to the InfiniTime commit log, the version bundled with InfiniTime 1.3.0  is nimble 1.3 master, commit 82153e744833821e20e9a8b0d61c38b2b0dbcfe1), I tried to update the included nimble to the most recent master - however, I was not able to successfully compile that combination yet.

Thank you for your continued help.

StarGate01, 2022-03-18:
I fixed the problem, I adjusted the file `nimble/drivers/nrf52/src/ble_phys.c` as follows:

- Replaced `NRF_CLOCK->TASKS_HFCLKSTOP = 1` with `nrf_drv_clock_hfclk_release()` in `ble_phy_rfclk_disable`
- Replaced `NRF_CLOCK->TASKS_HFCLKSTART = 1` with `nrf_drv_clock_hfclk_request()` in `ble_phy_rfclk_enable`

(In the Nimble upstream this is `nrf_clock_task_trigger`, but that function does the exact same thing)

The `nrf_drv_clock_hfclock_*` functions maintain an internal usage count to decide if the HF clock should be stopped or started, and since the HF clock is used for LF RC calibration as well I suspect that could have cause some contention issues.

I fixed this in my InfiniTime fork (see https://github.com/StarGate01/InfiniTime/commit/8878206c3a1646c85f111637f3138cce4369cb26). Since Nimble does not depend (I think) on the NRF SDK and its clock functions so far, I am hesitant to upstream this change. What do you think? How could the LFRC calibration / RFCLK-HFCLK contention be resolved?

andrzej-kaczmarek, 2022-03-20:
@StarGate01 we probably don't want to introduce dependency to nRF SDK, but recently we switched that code to use nrfx anyway (i.e. `nrf_clock_task_trigger` instead of using `NRF_CLOCK->TASK_...` directly) so if there's refcounting equivalent in nrfx for hfclk it can be used upstream

StarGate01, 2022-03-21:
@andrzej-kaczmarek Okay. It seems that the nrfx libs have no functionality to perform that reference counting, compare https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_clock.c and https://github.com/jamesmunns/nRF5-sdk/blob/master/components/drivers_nrf/clock/nrf_drv_clock.c .

I thought about checking `nrfx_clock_is_calibrating` before turning the hfclock off in  `ble_phy_rfclk_disable`, which would probably work. However, the calibration procedure could still turn off the hfclock once it is done, even though the bletooth stack is currently using it.

What do you think about using a weak symbol? Defining something like
```c++
__weak void nrf_drv_clock_hfclk_request(void * p_handler_item) {
    nrf_clock_task_trigger(NRF_CLOCK, NRF_CLOCK_TASK_HFCLKSTART);
}
```
should work (?)

The same would have to be done with `nrf_drv_clock_init`, which initializes the refcounter.

sjanc, 2022-04-13:
is this issues solved or something stil should be done on nimble side?

StarGate01, 2022-04-13:
I resolved the issue in the code I work on by applying the modification detailed above. However, that was in a custom fork of Nimble. My idea was to upstream this patch, even though I don't need it there anymore.