apache/mynewt-nimble #1213: app/blehci build fails

bvdberg, 2022-03-21:
Since the merge of the refactored transport layer, this build fails:

newt target create bb_test
newt target set bb_test build_profile=optimized
newt target set bb_test bsp=@apache-mynewt-core/hw/bsp/nordic_pca10095_net
newt target set bb_test app=@apache-mynewt-nimble/apps/blehci
newt build bb_test

Building target targets/bb_test
Error: Config for package @apache-mynewt-nimble/nimble/transport: setting BLE_ACL_BUF_SIZE does not have required value field

It seems to do with the import of "@apache-mynewt-nimble/nimble/transport/syscfg.defunct.yml"

sjanc, 2022-03-21:
Hi,

It seems to work for me...    please make sure you use newt from master

bvdberg, 2022-03-21:
That solved the issue, newt 1.9 -> 1.10.
Thx!

closed 2022-03-21 by bvdberg