apache/mynewt-nimble #1214: NRF5340 btshell build failing

bvdberg, 2022-03-21:
Before the new transport layer, I had the btshell working fine on a NRF5340 DK.
Now I'm following:
https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=177047756

In the syscfg.yml I changed BLE_HCI_TRANSPORT: nrf5340 into BLE_TRANSPORT_LL: nrf5340
It compiles, but linking  fails:

_newt build myapp
Building target targets/myapp
Compiling bin/targets/myapp/generated/src/myapp-sysinit-app.c
Archiving myapp-sysinit-app.a
Linking /home/bas/work/dev/newt/bin/targets/myapp/app/apps/myapp/myapp.elf
Error: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/bas/work/dev/newt/bin/targets/myapp/app/@apache-mynewt-nimble/nimble/transport/@apache-mynewt-nimble_nimble_transport.a(transport.o): in function `ble_transport_acl_put':
/home/bas/work/dev/newt/repos/apache-mynewt-nimble/nimble/transport/src/transport.c:177: undefined reference to `ble_transport_int_flow_ctl_put'
collect2: error: ld returned 1 exit status_

bvdberg, 2022-03-21:
I think this was caused by b44ec408820cb18c9d1ff4b6efac84322016d279 (nimble/transport: Add internal flow control)

sjanc, 2022-03-21:
you no longer specify HCI transport so you can just remove this sysval  (wiki might need to be updated to latest changes)

closed 2022-03-21 by bvdberg