apache/mynewt-nimble #1258: Bug: ".../nimble/host/include/host/ble_uuid.h:97:9: error: expected primary-expression before '.' token"

Kampi, 2022-05-13:
**Problem Description**

Using `static const ble_uuid128_t gatt_svr_svc_sec_test_uuid = BLE_UUID128_INIT(0x2d, 0x71, 0xa2, 0x59, 0xb4, 0x58, 0xc8, 0x12, 0x99, 0x99, 0x43, 0x95, 0x12, 0x2f, 0x46, 0x59);` from the NimBLE `bleprph` example will produce the error:

```
In file included from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs_adv.h:24,
                 from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:32,
                 from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:33,
                 from src/Bluetooth/bluetooth.cpp:9:
C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_uuid.h:97:9: error: expected primary-expression before '.' token
         .u.type = BLE_UUID_TYPE_128,    \
 ```
 
 **Reason**
 
 The macro uses the wrong structure format.

**Fix**

Change the macros to

```
#define BLE_UUID16_INIT(uuid16)         \
    {                                   \
        .u = {                          \
            .type = BLE_UUID_TYPE_16,   \
        },                              \
        .value = { uuid16 },            \
    }

#define BLE_UUID32_INIT(uuid32)         \
    {                                   \
        .u = {                          \
            .type = BLE_UUID_TYPE_32,   \
        },                              \
        .value = { uuid32 },            \
    }

#define BLE_UUID128_INIT(uuid128...)    \
    {                                   \
        .u = {                          \
            .type = BLE_UUID_TYPE_128,  \
        },                              \
        .value = { uuid128 },           \
    }
 ```

sjanc, 2022-05-16:
what toolchain is it?

sjanc, 2022-05-16:
but what compiler is used, GCC? clang?  and what version

Kampi, 2022-05-16:
It´s GCC and it should be version 8.4 if I´m correct.

Kampi, 2022-05-24:
Can be closed. The solution is accepted.

closed 2022-05-24 by Kampi