apache/mynewt-nimble #1264: NimBLE did not follow the sleep clock accuracy

sada45, 2022-05-17:
Hi!
    I am reading the Bluetooth spec v5.3. In the Vol 6, Part B, 4.5.7 "Sleep clock accuracy", the spec has claimed that the slave should update the its timing of the anchor point. However, it seems that the NimBLE only updates the timing at the first connection event. The corresponding code should be the `ble_ll_conn_rx_isr_start()` in `ble_ll_conn.c`.
[image]
`ble_ll_conn_rx_isr_start()` Line 3243:
```C
/* Set anchor point (and last) if 1st rxd frame in connection event */
if (connsm->csmflags.cfbit.periph_set_last_anchor) {
    connsm->csmflags.cfbit.periph_set_last_anchor = 0;
    connsm->last_anchor_point = rxhdr->beg_cputime;
    connsm->anchor_point = connsm->last_anchor_point;
    connsm->anchor_point_usecs = rxhdr->rem_usecs;
}
```

andrzej-kaczmarek, 2022-05-17:
This is exactly what this code does, i.e. updates anchor point on 1st PDU of each connection event. The `periph_set_last_anchor` flag is set for each connection event in `ble_ll_conn_event_start_cb`.

sada45, 2022-05-18:
Yes you are right. I did not understand the code correctly.

closed 2022-05-18 by sada45