apache/mynewt-nimble #1291: The connection subrating needs CONN_SUBRATING_HOST feature

sada45, 2022-06-27:
Hi!
I have tried the connection subrating in the Spec 5.3 and may found an issue.
Before the central sends the `SUBRATE_IND`, it will check wether the peripheral has the `BLE_LL_FEAT_CONN_SUBRATING_HOST`. The corresponding code is in `ble_ll.c` Line 4131. However, the peripheral will not initiate this feature, so the `ble_ll_conn_subrate_req_hci()` always return `-ENOTSUP`. 
A quick fix is modify the function `ble_ll_init()` at `ble_ll.c` Line 1965 to:
```C
#if MYNEWT_VAL(BLE_LL_CFG_FEAT_LL_ENHANCED_CONN_UPDATE)
    features |= BLE_LL_FEAT_CONN_SUBRATING;
    features |= BLE_LL_FEAT_CONN_SUBRATING_HOST;
#endif
```
By the way, it seems that there is no connection subrating API in the GAP layer, is this normal?

andrzej-kaczmarek, 2022-06-27:
Host supported features can be set by `HCI LE Set Host Feature`. NimBLE host does not have support for connection subrating so it does not set that bit.

sada45, 2022-07-09:
Thanks for your reply! I'll just use the connection subrating with link-layer APIs for now.

Here I found another question.
If the BLE works in the noisy RF environment, the keep-alive packet may be lost, and the link-layer chooses to retransmit the empty in the next connection event. 
However, when there is a data packet enqueued, the link-layer still choose to retransmit the empty packet, and the data packets will be stalled.
Is this policy defined in the Bluetooth spec. or necessary?

sada45, 2022-07-12:
I test the connection subrating but find the `Continuation Number` does not work.
I think I find the reason. In the function `ble_ll_conn_next_event()` in `ble_ll_conn.c`, Line 2275. The central and peripheral decide wether they should turn on radio on the next connection or the next subrate base event. 
Accroding to the Bluetooth spec. If the next connection event is not a base event and the connection event number since we last received a PDU is less than `Continuation Number`, nodes will turn on their radio in the next connection even, rather than the next base event.
So, to my understanding, the correct code should be:
```C
  if ((connsm->cont_num > 0) &&
     /* (connsm->event_cntr + 1 == connsm->subrate_base_event + */
      (connsm->event_cntr + 1 != connsm->subrate_base_event + 
                                 connsm->subrate_factor) &&
      (connsm->event_cntr - connsm->last_pdu_event < connsm->cont_num)) {
      next_is_subrated = 0;
  }
```
Looking forward to your reply :-) @andrzej-kaczmarek

sada45, 2022-07-13:
Besides, currentlly, the `connsm->last_pdu_event` will be updated once received a PDU no matter it is empty or not.
However, this value should be only be updated if there is data transmitted in current connection event.
The spec give the defination of continuation event in `Vol.6 PartB 4.5.1`:
_A continuation event is a connection event where, in at least one of the previous connContinuationNumber connection events (ignoring any before the last subrated connection event), at least one packet was transmitted or validly received containing a Link Layer PDU **with a non-zero Length field**._
The figure 4.41 in `Vol.6 PartB 4.5` gives an example for this:
[image]

andrzej-kaczmarek, 2022-07-13:
@sada45 good catch, thanks. I created a fix for both https://github.com/apache/mynewt-nimble/pull/1297.

I still need to validate those more carefully since apparently software we use to run qualification test cases do not implement those test cases properly and they pass in all cases...

andrzej-kaczmarek, 2022-07-13:
@sada45 I reworked continuation events handling a bit more, it should be easier to understand now and I believe it should now be handled properly. let me know how it works for you.

sada45, 2022-07-13:
@andrzej-kaczmarek I have some doubts that wether the `has_nonempty_pdu` should be set before or after the packet is acked.
For me, I update the `last_data_pdu` in `ble_ll_conn_rx_isr_end`, after the code checked the CRC and SN, which means this data PDU is successfully transmitted and acked.
However, the Spec. has not given a clear explanation about that.

sada45, 2022-07-14:
Yes, that sounds reasonable to me.