apache/mynewt-nimble #1306: Example of starting and stopping ble

alberk8, 2022-07-25:
Is there any example of ble server/peripheral where the ble is stated, broadcast for a duration eg 90 seconds, then disposed/shutdown ble. Wait for 5 minutes create start broadcast again in this cycle.

alberk8, 2022-07-27:
Can someone please take a look at this implementation,  it looks like the deinit does not release the `TaskHandle_t host_task_h` [    nimble_port_freertos_deinit()](https://github.com/espressif/esp-nimble/blob/dac3a2a0f54589b10a4cff2a99526804747089df/porting/npl/freertos/src/nimble_port_freertos.c#L54)

bvdberg, 2022-08-17:
You should set host_task_h to NULL inside the if statement. But Nimble also uses Timers and possibly a 2nd Task to run, so you should stop those as well.