apache/mynewt-nimble #1315: Documentation: Assist in discovering ports

chrysn, 2022-08-04:
nimble is portable both with respect to the operating system it runs on and to the radio modules, but starting at the project web page or repository it is hard to discover for which there is which level of support.

I suggest that that the Supported Hardware section of the README be extended, eg. like this:

> With the components in this repository, NimBLE supports Nordic nRF51 and nRF52 chipsets as controllers. Host runs on any board and architecture [supported](https://github.com/apache/mynewt-core#overview) by Apache Mynewt OS.
>
> Several other projects provide out-of-tree support for additional components:
> * The [Espressif ESP-IDF contains a NimBLE port](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html) for ESP-32 devices.
> * The [RIOT operating system contains a package](https://doc.riot-os.org/group__pkg__nimble.html) for using NimBLE.
> * *If you publish a NimBLE port, please let us know to include it here!*

I think that the ESP-IDF stuff is not really a port but more of a fork, but don't know it well enough to make a good assessment.

The porting/targets directory contains snippets for Linux and nuttx, so they should probably also get mentions on that list if they're usable.

That information might also be maintained, for example, in a wiki, but I think it's practical to have this highly visible.

sjanc, 2022-09-22:
Something like this maybe? https://github.com/apache/mynewt-nimble/pull/1366

chrysn, 2022-09-22:
Yeah, that looks good, thanks.

Is there any hardware support that resides out-of-tree? Or is there a current list of in-tree supported hardware to reference? (It currently says nRF51 and nRF52 in the docs, but recent PRs make it look like nRF53 works too).

sjanc, 2022-09-22:
Bluetooth host is CPU agnostic, for controller we support Nordic (nRF51, nRF52, nRF5340) and Dialog DA1469x (dialog_cmac)  (https://github.com/apache/mynewt-nimble/tree/master/nimble/drivers for reference)

sjanc, 2024-02-01:
I guess this can be closed

closed 2024-02-01 by sjanc