apache/mynewt-nimble #1416: LOG_LEVEL defines collision

vovagorodok, 2022-12-01:
At `log_common.h` we have defined:
```
#define LOG_LEVEL_DEBUG    (0)
#define LOG_LEVEL_INFO     (1)
#define LOG_LEVEL_WARN     (2)
#define LOG_LEVEL_ERROR    (3)
#define LOG_LEVEL_CRITICAL (4)
#define LOG_LEVEL_NONE     (5)
```
`LOG_LEVEL..` names are very common and I have redefinition issues when another than NibBle logger is used.\
Because its internal NibBle logger, proposed to use some internal names like:
`BLE_LOG_LEVEL..` or `NIM_BLE_LOG_LEVEL..`

vovagorodok, 2023-01-07:
What do You think?

sjanc, 2023-01-27:
Hi,

Currently logs implementation is from  Mynewt but we will extend NPL with logs so that this is no longer the case.
I'm bit busy at the moment but will try to find some time in upcoming weeks to work on this.
Ping me if this is not done i next few weeks :)

vovagorodok, 2023-03-05:
Ping :)

vovagorodok, 2023-03-22:
@sjanc found time to rename it?

vovagorodok, 2023-07-18:
@sjanc I'm back with this topic

sjanc, 2023-10-06:
should be fixed upstream, sorry it took a bit longer than expected

closed 2023-10-06 by sjanc

vovagorodok, 2023-10-06:
Nice :) Thanks