apache/mynewt-nimble #1421: When using COC server and send is stalled, disconnects are not detected

bvdberg, 2022-12-14:
We have an NRF5340 device running Nimble Peripheral. It has a COC server and a GATT server. We connect from a PC application and the device starts streaming data to the PC. Sometimes we get a stalled, then unstalled and everything continues. But sometimes we get a stalled and the PC side gets a disconnect (Connection time-out). The Nimble side doesn't give a GAP disconnect event and just nothing. It never detects the disconnect. We tried manually disconnecting (using both *ble_l2cap_disconnect* and *ble_gap_terminate* (both return 0), but nothing happens. Maybe some bottom layer/queue is still locked?

rymanluk, 2023-01-19:
Do you possibly have any logs to share?

bvdberg, 2023-01-20:
Nothing is logged in the device. The PC side detects a connection drop (and starts trying to re-connect). The Device (Nimble) side doesn't log anything special. The device is still running (blinking LEDs etc). Could it be that it runs out of a command buffer and cannot get a disconnect message up?