apache/mynewt-nimble #1438: NimBle as host only HCI Uart

suadsuljic7, 2023-01-27:
Hi,
I'm trying to use NimBle as host only to communicate over HCI Uart with an nrf52 Bluetooth.
So I have followed this https://github.com/apache/mynewt-nimble/pull/1399.
Pulled the changes, add the BLE_TRANSPORT_LL: uart_ll in syscfg.yml from my app.
But when I try to build the app I m getting the following error:
[image]

sjanc, 2023-07-04:
should be fixed in master now

closed 2023-07-04 by sjanc

bojanpotocnik, 2024-05-13:
Today I am getting the same error
```yaml
syscfg.vals:
    # Interface external HCI controller using UART H4 transport protocol:
    BLE_TRANSPORT_HS: native
    BLE_TRANSPORT_LL: uart_ll
    BLE_TRANSPORT_UART_PORT: 0
    BLE_TRANSPORT_UART_BAUDRATE: 115200
    BLE_TRANSPORT_UART_FLOW_CONTROL: off
```
```
$ newt version
Apache Newt 1.12.0-dev / unknown / unknown

$ newt upgrade
Skipping "apache-mynewt-core": already upgraded (0.0.0)
Skipping "apache-mynewt-mcumgr": already upgraded (0.0.0)
Skipping "apache-mynewt-nimble": already upgraded (0.0.0)
Skipping "mcuboot": already upgraded (2.1.0)
Skipping "apache-mynewt-core": already upgraded (0.0.0)
Skipping "apache-mynewt-mcumgr": already upgraded (0.0.0)
Skipping "apache-mynewt-nimble": already upgraded (0.0.0)
Skipping "arm-CMSIS_5": already upgraded (5.4.0)
Skipping "atmel-samd21xx": already upgraded (mynewt_1_10_0_tag)
Skipping "eembc-coremark": already upgraded (v1.01)
Skipping "liblc3": already upgraded (v1.0.4)
Skipping "libsamplerate": already upgraded (0.2.2)
Skipping "littlefs": already upgraded (v2.5.0)
Skipping "lvgl": already upgraded (v8.3.6)
Skipping "mbedtls": already upgraded (v2.28.8)
Skipping "mcuboot": already upgraded (2.1.0)
Skipping "nordic-nrfx": already upgraded (v3.3.0)
Skipping "stm-cmsis_device_f0": already upgraded (v2.3.7)
Skipping "stm-cmsis_device_f1": already upgraded (v4.3.3)
Skipping "stm-cmsis_device_f3": already upgraded (v2.3.7)
Skipping "stm-cmsis_device_f4": already upgraded (v2.6.8)
Skipping "stm-cmsis_device_f7": already upgraded (v1.2.8)
Skipping "stm-cmsis_device_g0": already upgraded (v1.4.3)
Skipping "stm-cmsis_device_g4": already upgraded (v1.2.0)
Skipping "stm-cmsis_device_h7": already upgraded (v1.10.3)
Skipping "stm-cmsis_device_l0": already upgraded (v1.9.3)
Skipping "stm-cmsis_device_l1": already upgraded (v2.3.2)
Skipping "stm-cmsis_device_l4": already upgraded (v1.7.2)
Skipping "stm-cmsis_device_u5": already upgraded (v1.2.0)
Skipping "stm-cmsis_device_wb": already upgraded (v1.12.0)
Skipping "stm-stm32f0xx_hal_driver": already upgraded (v1.7.7)
Skipping "stm-stm32f1xx_hal_driver": already upgraded (v1.1.8)
Skipping "stm-stm32f3xx_hal_driver": already upgraded (v1.5.7)
Skipping "stm-stm32f4xx_hal_driver": already upgraded (v1.7.13)
Skipping "stm-stm32f7xx_hal_driver": already upgraded (v1.2.10)
Skipping "stm-stm32g0xx_hal_driver": already upgraded (v1.4.5)
Skipping "stm-stm32g4xx_hal_driver": already upgraded (v1.2.0)
Skipping "stm-stm32h7xx_hal_driver": already upgraded (v1.11.1)
Skipping "stm-stm32l0xx_hal_driver": already upgraded (v1.10.6)
Skipping "stm-stm32l1xx_hal_driver": already upgraded (v1.4.4)
Skipping "stm-stm32l4xx_hal_driver": already upgraded (v1.13.3)
Skipping "stm-stm32u5xx_hal_driver": already upgraded (v1.2.0)
Skipping "stm-stm32wbxx_hal_driver": already upgraded (v1.13.0)
Skipping "tinyusb": already upgraded (0.15.0)
```
results in 
```
Compiling repos/apache-mynewt-core/sys/log/full/src/log_fcb2.c
Error: In file included from repos/apache-mynewt-core/sys/sysinit/include/sysinit/sysinit.h:25,
                 from repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c:24:
repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c: In function 'hci_uart_configure':
bin/targets/btshell_native/generated/include/syscfg/syscfg.h:15:49: error: 'MYNEWT_VAL_BLE_TRANSPORT_UART_PORT' undeclared (first use in this function); did you mean 'MYNEWT_VAL_BLE_TRANSPORT_UART_LL_PORT'?
   15 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c:139:28: note: in expansion of macro 'MYNEWT_VAL'
[... 47 lines cut ...]
      |                                                 ^~~~~~~~~~~
repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c:195:23: note: in expansion of macro 'MYNEWT_VAL'
  195 |     hal_uart_start_tx(MYNEWT_VAL(BLE_TRANSPORT_UART_PORT));
      |                       ^~~~~~~~~~
repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c: In function 'ble_transport_to_ll_acl_impl':
bin/targets/btshell_native/generated/include/syscfg/syscfg.h:15:49: error: 'MYNEWT_VAL_BLE_TRANSPORT_UART_PORT' undeclared (first use in this function); did you mean 'MYNEWT_VAL_BLE_TRANSPORT_UART_LL_PORT'?
   15 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
repos/apache-mynewt-nimble/nimble/transport/uart_ll/src/hci_uart.c:223:23: note: in expansion of macro 'MYNEWT_VAL'
  223 |     hal_uart_start_tx(MYNEWT_VAL(BLE_TRANSPORT_UART_PORT));
      |                       ^~~~~~~~~~
```
I thought this is using latest master... should I set some specifics somewhere? Using `vers: 1-latest` instead of `0-dev` or `0.0.0` results in
```
$ newt upgrade
Making the following changes to the project:
    upgrade apache-mynewt-core (0.0.0 --> 1.12.0)
    upgrade apache-mynewt-mcumgr (0.0.0 --> 0.4.0)
    upgrade apache-mynewt-nimble (0.0.0 --> 1.7.0)
    upgrade mcuboot (2.1.0 --> 2.0.0)
Error: This version of newt (1.11.9900) is incompatible with your version of the apache-mynewt-core repo (1.12.0); Please upgrade your newt tool to version 1.12.0
This version of newt (1.11.9900) is incompatible with your version of the apache-mynewt-nimble repo (1.7.0); Please upgrade your newt tool to version 1.12.0
```

Thank you for any help :pray:

bojanpotocnik, 2024-05-13:
"Upgraded" newt to 1.12.0 and project to `vers: 1-latest`:
```
$ newt upgrade
Making the following changes to the project:
    upgrade apache-mynewt-core (0.0.0 --> 1.12.0)
    upgrade apache-mynewt-mcumgr (0.0.0 --> 0.4.0)
    upgrade apache-mynewt-nimble (0.0.0 --> 1.7.0)
    upgrade mcuboot (2.1.0 --> 2.0.0)
apache-mynewt-core successfully upgraded to version 1.12.0
apache-mynewt-mcumgr successfully upgraded to version 0.4.0
apache-mynewt-nimble successfully upgraded to version 1.7.0
mcuboot successfully upgraded to version 2.0.0
Skipping "apache-mynewt-core": already upgraded (0.0.0)
Skipping "arm-CMSIS_5": already upgraded (5.4.0)
Skipping "atmel-samd21xx": already upgraded (mynewt_1_10_0_tag)
Skipping "liblc3": already upgraded (v1.0.4)
Skipping "libsamplerate": already upgraded (0.2.2)
Skipping "littlefs": already upgraded (v2.5.0)
Skipping "lvgl": already upgraded (v8.3.6)
Skipping "nordic-nrfx": already upgraded (v3.3.0)
Skipping "stm-cmsis_device_f0": already upgraded (v2.3.7)
Skipping "stm-cmsis_device_f1": already upgraded (v4.3.3)
Skipping "stm-cmsis_device_f3": already upgraded (v2.3.7)
Skipping "stm-cmsis_device_f4": already upgraded (v2.6.8)
Skipping "stm-cmsis_device_f7": already upgraded (v1.2.8)
Skipping "stm-cmsis_device_g0": already upgraded (v1.4.3)
Skipping "stm-cmsis_device_g4": already upgraded (v1.2.0)
Skipping "stm-cmsis_device_h7": already upgraded (v1.10.3)
Skipping "stm-cmsis_device_l0": already upgraded (v1.9.3)
Skipping "stm-cmsis_device_l1": already upgraded (v2.3.2)
Skipping "stm-cmsis_device_l4": already upgraded (v1.7.2)
Skipping "stm-cmsis_device_u5": already upgraded (v1.2.0)
Skipping "stm-cmsis_device_wb": already upgraded (v1.12.0)
Skipping "stm-stm32f0xx_hal_driver": already upgraded (v1.7.7)
Skipping "stm-stm32f1xx_hal_driver": already upgraded (v1.1.8)
Skipping "stm-stm32f3xx_hal_driver": already upgraded (v1.5.7)
Skipping "stm-stm32f4xx_hal_driver": already upgraded (v1.7.13)
Skipping "stm-stm32f7xx_hal_driver": already upgraded (v1.2.10)
Skipping "stm-stm32g0xx_hal_driver": already upgraded (v1.4.5)
Skipping "stm-stm32g4xx_hal_driver": already upgraded (v1.2.0)
Skipping "stm-stm32h7xx_hal_driver": already upgraded (v1.11.1)
Skipping "stm-stm32l0xx_hal_driver": already upgraded (v1.10.6)
Skipping "stm-stm32l1xx_hal_driver": already upgraded (v1.4.4)
Skipping "stm-stm32l4xx_hal_driver": already upgraded (v1.13.3)
Skipping "stm-stm32u5xx_hal_driver": already upgraded (v1.2.0)
Skipping "stm-stm32wbxx_hal_driver": already upgraded (v1.13.0)
Skipping "tinyusb": already upgraded (0.15.0)
Making the following changes to the project:
    upgrade apache-mynewt-mcumgr (0.0.0 --> 0.4.0)
    upgrade apache-mynewt-nimble (0.0.0 --> 1.7.0)
    upgrade mbedtls (5a764e5555c64337ed17444410269ff21cb617b1 --> v2.28.4)
    upgrade mcuboot (2.1.0 --> 2.0.0)
apache-mynewt-mcumgr successfully upgraded to version 0.4.0
apache-mynewt-nimble successfully upgraded to version 1.7.0
mbedtls successfully upgraded to version v2.28.4
mcuboot successfully upgraded to version 2.0.0
```
but the error is still the same.