apache/mynewt-nimble #1463: porting/nimble/Makefile.defs C and LD flags issue

nikitos1550, 2023-02-17:
Hello! 
There is such lines in `porting/nimble/Makefile.defs`:
```
[CUT]

# For now this is required as there are places in NimBLE
# assumingthat pointer is 4 bytes long.
NIMBLE_CFLAGS := -m32
NIMBLE_LDFLAGS := -m32
[CUT]
```

I am cross compiling sample linux application for ARM Cortex-A7 and my toolchain does not recognize `-m32` option, as it is already 32bit (I know about nimble 64bit issue).

Is it possible to change `porting/nimble/Makefile.defs` with something like this:
```
NIMBLE_CFLAGS ?= -m32
NIMBLE_LDFLAGS ?= -m32
```
then it will follow same behavior as before, but I can override these vars in my application Makefile.
Of course I can patch nimble sources, but It will be nice if it support such _customization_ out of the box.

nikitos1550, 2023-02-17:
Also I realized that I can simply remove usage for this vars in my local Makefile, but anyway I need somebody comment on this.

sjanc, 2023-02-21:
Initially that host port was only used on x86 linux and thus -m32.    I'm fine with improving makefiles to cover also other usecases.  But note that running on 64bit may require reviewing nimble host code for assumptions that pointer is 32bit long

closed 2023-04-23 by nikitos1550