apache/mynewt-nimble #1488: extended discovery, phy 2Mbit, comment needed

nikitos1550, 2023-04-23:
Hello!

I am trying use ble in 2mbit mode (actually I am not too strong in this area right now and maybe do something wrong)

```
[CUT]
    struct ble_gap_ext_disc_params params = {0};

    params.itvl = BLE_GAP_SCAN_FAST_INTERVAL_MAX;
    params.passive = 1;
    params.window = BLE_GAP_SCAN_FAST_WINDOW;

    rc = ble_gap_ext_disc(own_addr_type, 3*1000, 0, 1, 0, 0,
                     NULL /*&params*/ /*const struct ble_gap_ext_disc_params *uncoded_params*/,
                     &params /*NULL*/ /*const struct ble_gap_ext_disc_params *coded_params*/,
                     nimble_binding_gap_event_cb, NULL);
    if (rc != 0) {
        fprintf(stderr, "%s: Error initiating GAP discovery procedure; rc=%d\n", __func__,
                    rc);
    }
[CUT]
```
But I got rc=8 (NOT SUPPORTED).  
I suppose that my ble usb dongle just does not support it, but I am not sure.
I have some different dongles (here I log list of all adapters that are available for me https://github.com/nikitos1550/ble-notes/tree/main/usb-adapters). How can I check that my dongle support it? Maybe there are some dependency on exact BLE version (5.0, 5.1, 5.2 , 5.3 ???).

bvdberg, 2023-05-12:
Did you enable: MYNEWT_VAL_BLE_LL_CFG_FEAT_LE_2M_PHY ?

nikitos1550, 2023-05-15:
No, there is no such thing in my config, I will check. Thanks!

sjanc, 2023-07-04:
Hi, if this is stil an issue please re-open

closed 2023-07-04 by sjanc