apache/mynewt-nimble #1550: Linux unit test: Segmentation fault

pkarashchenko, 2023-07-16:
Preconditions: hash `98d972d220a0a49f40dca8074418397d5323d1ec` (`master`)
Steps to reproduce:
1. `cd porting/npl/linux/test`
2. `make`
3. `make test`
Result:
```
$ make test
./test_npl_task.exe
All tests passed
./test_npl_callout.exe
All tests passed
./test_npl_eventq.exe
All tests passed
./test_npl_sem.exe
make: *** [Makefile:94: test] Segmentation fault (core dumped)
```

Also some warnings related to `-Wimplicit-function-declaration` observed during compilation.

Initially my goal was to prototype BLE central application on Linux, but in a simple code I'm getting `BLE_GAP_EVENT_DISC` with `struct ble_gap_disc_desc`->`event_type` values of 114 or 144 or `146` (and more random data), so I switched to unit tests. I will try out some stable release.

pkarashchenko, 2023-07-16:
Root cause seems to be a missing prototype of `ble_npl_eventq_dflt_get` as after I add a prototype the test pass

pkarashchenko, 2023-07-16:
Also there is an error when running `make clean`: `rm: cannot remove '*~': No such file or directory`

pkarashchenko, 2023-07-16:
Also unit test section from `porting/examples/linux/README.md` does not reflect the actual steps to build and run unit tests.

sjanc, 2023-07-17:
Hi,

Hmm those seem to pass on my box (fedora linux)...

Please send PR with changes that fix this for you.
I also noticed that those tests are not executed in github actions (just build test), I'll add them

sjanc, 2023-07-20:
https://github.com/apache/mynewt-nimble/pull/1565  should fix 'make clean' error and also adds Linux NPL test to CI workflows

sjanc, 2023-07-20:
interestingly it fails on CI with same error as this issue :)

pkarashchenko, 2023-07-20:
Maybe because `-m32` option is added for the code build, but not added for tests?
Sorry, I've been busy and didn't find a time to submit a fix of missing prototypes.

sjanc, 2023-07-20:
yeap, I'll enable -m32 for now and later on we can extend it for 64bits

closed 2023-07-21 by pkarashchenko