apache/mynewt-nimble #1551: Runnig of blecent app on Linux

pkarashchenko, 2023-07-16:
I tried to run `blecent` app on Linux. To do this I copied corresponding source files from `apps/blecent` to `porting/examples/linux` and adapted `Makefile` and `blecent` code to build and run. I also added `printf("event->disc.event_type %d\n", event->disc.event_type);` print to `case BLE_GAP_EVENT_DISC:` in `blecent_gap_event`. After I run the program I' observing next output
```
$ sudo ./nimble-linux 
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1 duration=forever
event->disc.event_type 128
    flags=0x06
    uuids128(complete)=d846fc4a-84cc-4c56-9786-187b0725751d 
    appearance=0x0000
event->disc.event_type 18
    uuids16(complete)=0xfef3
```
but `event_type` is stated to carry
```
    /** Advertising PDU type. Can be one of following constants:
     *  - BLE_HCI_ADV_RPT_EVTYPE_ADV_IND
     *  - BLE_HCI_ADV_RPT_EVTYPE_DIR_IND
     *  - BLE_HCI_ADV_RPT_EVTYPE_SCAN_IND
     *  - BLE_HCI_ADV_RPT_EVTYPE_NONCONN_IND
     *  - BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP
     */
    uint8_t event_type;
```
values from `0` to `4`.

pkarashchenko, 2023-07-16:
The unknown event types seems to be reported by Linux HCI socket interface:
[image]

I did a quick experiment and see that only lower 4 bits seems to be valid and changing code:
```
diff --git a/nimble/host/src/ble_hs_hci_evt.c b/nimble/host/src/ble_hs_hci_evt.c
index e7af2243..3096b4fc 100644
--- a/nimble/host/src/ble_hs_hci_evt.c
+++ b/nimble/host/src/ble_hs_hci_evt.c
@@ -481,7 +481,7 @@ ble_hs_hci_evt_le_adv_rpt(uint8_t subevent, const void *data, unsigned int len)
 
         data += sizeof(rpt) + rpt->data_len + 1;
 
-        desc.event_type = rpt->type;
+        desc.event_type = rpt->type & 0xF;
         desc.addr.type = rpt->addr_type;
         memcpy(desc.addr.val, rpt->addr, BLE_DEV_ADDR_LEN);
         desc.length_data = rpt->data_len;
@@ -510,7 +510,7 @@ ble_hs_hci_evt_le_dir_adv_rpt(uint8_t subevent, const void *data, unsigned int l
     desc.length_data = 0;
 
     for (i = 0; i < ev->num_reports; i++) {
-        desc.event_type = ev->reports[i].type;
+        desc.event_type = ev->reports[i].type & 0xF;
         desc.addr.type = ev->reports[i].addr_type;
         memcpy(desc.addr.val, ev->reports[i].addr, BLE_DEV_ADDR_LEN);
         desc.direct_addr.type = ev->reports[i].dir_addr_type;
```
seems to solve the issue. I'm not sure if that is a correct solution

sjanc, 2023-07-17:
Hi,

this is odd, 0xa1 doesn't even match report type from extended advertising report...   what controller is this?

As for the patch,  it looks OK, please send PR (we may discuss if this should be under some ifdef ie, for non-mynewt controllers only, but this can happen in PR itself)

pkarashchenko, 2023-07-17:
I will check the controller model. Actually in this case I'm running Ubuntu 20.04 on MacBook Pro :)
I will get the controller details and also will check if this is a driver issue. Also I will check if the same happens if I run exactly the same OS (installed on a portable SSD) on a Dell laptop.

pkarashchenko, 2023-07-20:
Seems to be Bluetooth driver instability when running Ubuntu on MacBook.

closed 2023-07-20 by pkarashchenko