apache/mynewt-nimble #1662: SC Pairing not working: Inconsistent behavior in DH key exchange.

sacca97, 2023-12-15:
When attempting to pair using SC with JustWorks inside the `ble_sm_sc_public_key_exec()` function the call to `ble_sm_tx()` writes the wrong content to the output buffer ~80% of the time, causing SC Pairing to fail. (testing on peripheral)

This only seems to happen in this specific case when the packet is bigger than a certain size (the pubkey is 64) and explains why it is the only function that has this problem.

While debugging I found that the buffer content changes already in the first line of `ble_sm_tx()`, so technically before any other operation happens. 

I cannot figure out why this is happening but as temporary solution I am copying the data in the `om_buf` by using the `_copyinto` function.

sacca97, 2023-12-18:
My hypothesis: this is due to how the m_bufs are managed.
So if you try to fill them just by working with pointers you will break something (?) if the content is too big.
Would like some feedback on this, there's a completely broken Secure Connection.

sjanc, 2023-12-18:
Hi,

Do you tune MSYS_1_BLOCK_SIZE ?  Default 292 should be fine for LE SC.
I think SMP code assumes continuous buffer for SMP commands so probably we miss some restrictions regarding this (at least as short term solution)

sacca97, 2023-12-18:
Yeah I was using the provided sample app bleprp, which sets it to 110. I can see the problem now.
I can work on a fix that fits all the scenarios in the next days and make a PR.

rymanluk, 2024-05-07:
Hi @sacca97,  when do you plan to land the PR ? :)

sacca97, 2024-05-22:
Hey I've been a bit overwhelmend and also forgot abt it. I have made some changes in my fork but could not test it properly yet (it works in the case of the MSYS_1_BLOCK_SIZE=110 as in the blehr app at least).
It is on my profile if you want to check it.