apache/mynewt-nimble #1669: nimble-linux: ble_hci_sock_init: Assertion `rc == 0' failed.

Fanhang32, 2023-12-30:
The linux examples failed to run in the latest code
```
bind() failed 16 hci0
nimble-linux: ../../../nimble/transport/socket/src/ble_hci_socket.c:828: ble_hci_sock_init: Assertion `rc == 0' failed.
Aborted
```
The root cause was `ble_hci_sock_init` duplicated called in `ble_transport_ll_init` and linux example code. 
It was introduced by this PR https://github.com/apache/mynewt-nimble/pull/1597.

@sjanc @CW-B-W Could you tell me why we need to initialize `ble_transport_ll_init` when use external controller?

Fanhang32, 2023-12-30:
If it is indeed necessary to do so，i can fix it by remove `ble_hci_sock_init ` from `ble_transport_ll_init ` in `ble_hci_socket.c`

andrzej-kaczmarek, 2023-12-30:
`ble_hci_sock_init` calls should be removed from apps.

`ble_transport_ll_init` initializes LL side of transport which in case of external controller is a HCI socket so the socket should be initialized there.

CW-B-W, 2023-12-30:
Hello @Fanhang32,

I'm only a user of this project, and I only played it with FreeRTOS, therefore the below might not be completely correct.

---

In **mynewt-nimble v1.5.0**, the architecture of HCI transport had been modified.
(You may refer to [nimble/doc/transport.md](https://github.com/apache/mynewt-nimble/blob/master/nimble/doc/transport.md) for the current architecture.)

When I was migrating my personal project from **v1.4.0** to **v1.5.0**, I found the examples at that time were not compatible with **v1.5.0**.
For adapting to the v1.5.0 architecture, I opened PR https://github.com/apache/mynewt-nimble/pull/1597 to resolve the issue.

Is the example you used [porting/examples/linux](https://github.com/apache/mynewt-nimble/blob/master/porting/examples/linux/main.c)?

As I saw [porting/examples/linux](https://github.com/apache/mynewt-nimble/blob/master/porting/examples/linux/main.c) has not been maintained for a long time, I think this example might be kinda outdated and need to be adapted to the new HCI transport architecture.

Would it be better if we keep `ble_hci_sock_init()` in `ble_transport_ll_init()`
https://github.com/apache/mynewt-nimble/blob/64067f5168bd4de097b7830a078862a5df95d9a5/nimble/transport/socket/src/ble_hci_socket.c#L836-L841

and the remove `ble_hci_sock_init()` in the example (`porting/examples/linux`)?
https://github.com/apache/mynewt-nimble/blob/64067f5168bd4de097b7830a078862a5df95d9a5/porting/examples/linux/main.c#L69-L70

I think `ble_transport_ll_init()` is a more upper-level API for users to initialize the controller, while `ble_hci_socket_init()` may be the platform-specific implementation for `ble_transport_ll_init()`?

Fanhang32, 2023-12-31:
@andrzej-kaczmarek @CW-B-W Got it , thanks a lot！ I‘ve removed `ble_hci_sock_init` in example apps in https://github.com/apache/mynewt-nimble/pull/1670

closed 2023-12-31 by andrzej-kaczmarek