apache/mynewt-nimble #1717: Error code of ble_gap_adv_set_data function

taks, 2024-03-02:
The `ble_gap_adv_set_data` function returns `BLE_ERR_INV_HCI_CMD_PARMS` if the argument is invalid.
https://github.com/apache/mynewt-nimble/blob/a04a6d2ce642bb0f7f49bf60c8b37988152548c0/nimble/host/src/ble_gap.c#L2772-L2776

However, other functions (e.g., `ble_gap_adv_rsp_set_data`) return `BLE_HS_EINVAL`.
https://github.com/apache/mynewt-nimble/blob/a04a6d2ce642bb0f7f49bf60c8b37988152548c0/nimble/host/src/ble_gap.c#L2800-L2804

Is there any reason why the `ble_gap_adv_set_data` function returns `BLE_ERR_INV_HCI_CMD_PARMS` instead of `BLE_HS_EINVAL`?

andrzej-kaczmarek, 2024-03-03:
no, it should return `BLE_HS_INVAL`. could you please create a PR to fix this issue?

taks, 2024-03-04:
Thank you for your confirmation.
I have created a PR.

PR #1718 (2024-03-05): nimble/host: Fix ble_gap_adv_set_data return code
Fix #1717
closed 2024-03-05 by sjanc