apache/mynewt-nimble #1800: host: assertion in ble_eatt_tx

piotrnarajowski, 2024-06-25:
While going through GATT/SR/GAW/BV-10-C test case (and several other GATT/SR test cases), an failed assertion occurs in function **ble_eatt_tx()** after PTS sends ATT_READ_REQ. This is because **ble_l2cap_send()** returns **BLE_HS_EBADDATA** error (SDU length is greater than the peer device's MTU). For some reason this doesn't happen every time. 

    ble_eatt_tx(uint16_t conn_handle, uint16_t cid, struct os_mbuf *txom)
    {
        struct ble_eatt *eatt;
        volatile int rc;

        BLE_EATT_LOG_DEBUG("eatt: %s, size %d ", __func__, OS_MBUF_PKTLEN(txom));
        eatt = ble_eatt_find(conn_handle, cid);
        if (!eatt || !eatt->chan) {
            BLE_EATT_LOG_ERROR("Eatt not available");
            rc = BLE_HS_ENOENT;
            goto error;
        }

        rc = ble_l2cap_send(eatt->chan, txom);
        if (rc == 0) {
            goto done;
        }

        if (rc == BLE_HS_ESTALLED) {
            BLE_EATT_LOG_DEBUG("ble_eatt_tx: Eatt stalled");
        } else if (rc == BLE_HS_EBUSY) {
            BLE_EATT_LOG_DEBUG("ble_eatt_tx: Message queued");
            STAILQ_INSERT_HEAD(&eatt->eatt_tx_q, OS_MBUF_PKTHDR(txom), omp_next);
            ble_npl_eventq_put(ble_hs_evq_get(), &eatt->wakeup_ev);
        } else {
            BLE_EATT_LOG_ERROR("eatt: %s, ERROR %d ", __func__, rc);
            assert(0);
        }
    done:
    return 0;

    error:
        os_mbuf_free_chain(txom);
        return rc;
    }

piotrnarajowski, 2024-06-25:
FYI @rymanluk

KKopyscinski, 2024-06-26:
I'm not sure why we need to send SDU greater then peer's MTU, which is not according to spec (3.4.2 L2CAP SDU Length field (2 octets)). But the second thing is, does this really need to hit assert? Maybe we should add clause for `BLE_HS_EBADDATA`, log it and `goto error`

piotrnarajowski, 2024-06-26:
this might be useful [GATT_SR_GAW_BV-10-C_2024_06_26_12_08_01.zip](https://github.com/user-attachments/files/15986837/GATT_SR_GAW_BV-10-C_2024_06_26_12_08_01.zip)

piotrnarajowski, 2024-09-24:
Fixed https://github.com/apache/mynewt-nimble/pull/1868

closed 2024-09-24 by piotrnarajowski