apache/mynewt-nimble #1827: Error status 0x0181 sent back in BLE write call-back.  What does it mean?

elfrances, 2024-08-05:
My app is getting back the error status 0x0181 (385 decimal) in the BLE write callback when it tries to write to the FTMS's Fitness Machine Control Point characteristic 0x2AD9.  Browsing the NimBLE code, I can't figure out what this status code means.  The BLE device rejecting this write request is a ZwiftHub indoor bike trainer.  No idea what BLE stack this device uses.  Does anyone know?

sjanc, 2024-08-06:
Hi,

In Nimble error codes have ranges so that one can figure out from which layer error originated from.
0x181 is 0x100 + 0x81 which means it is ATT (0x100) error code with value 0x81.

0x81 is from ATT application error code range which means that its meaning is defined in FTMS specification.

More info on errors can be found in documentation: https://mynewt.apache.org/v1_12_0/network/ble_hs/ble_hs_return_codes.html

elfrances, 2024-08-06:
Thanks for your clear explanation @sjanc !

For NimBLE neophytes (such as me) it would be helpful to have a comment with your clear explanation in the definition of the 'struct ble_gatt_error' inside ble_gatt.h.

closed 2026-09-08 by sjanc