apache/mynewt-nimble #1865: using namespace std; in Header File breaks compilation

timoxd7, 2024-09-17:
## Info
After updating our projects ESP IDF Version, which also updates the NimBLE Stack, the Code was not able to compile anymore.

One Reason was this line:

https://github.com/apache/mynewt-nimble/blob/16e1bfc67966ba531d1c313f443ced7f66a2e15e/porting/npl/nuttx/src/wqueue.h#L37

Every File including this directly or indirectly will be forced to use the `std` namespace, which should _not_ be. Removing this line works for me, however i don't know if it is conditionally needed by other platforms. However, this should never appear in a header file and if needed, must be moved to the according c/cpp file or using the `std::` syntax in front of necessary functions.

sjanc, 2024-09-17:
Hi,

While I agree about namespaces in headers,  this header is internal (in src/ folder, and probably not even used in nuttx port, probably copy-and-paste from linux port which has this already fixed)

I just wonder how this is affecting your build in first place ?

sjanc, 2024-09-17:
FYI https://github.com/apache/mynewt-nimble/pull/1866

timoxd7, 2024-09-17:
@sjanc Thanks, just had another thing (min/max definition) which prevented my build, might be that the `using namespace std` was not the root causing issue. The min/max problem is already fixed in NimBLE, but not in the ESP IDF Port (i think i made a PR for this a while ago which was merged into NimBLE but not ESP IDF). Will close this issue.

closed 2024-09-17 by timoxd7