apache/mynewt-nimble #1925: Improve latency performance with better retransmittion mechanism

sada45, 2024-11-25:
Hi!

I think there is a possible method to improve the uploading (Peripheral to Central) transmission latency.
BLE specification set if there are two continue packet loss then the connection event should be terminated.
However, consider that while the Peripheral sending multiple PDUs to the Central while the Central has no data to transit.
If one packet (expect the last one) is lost, the Central will not reply an empty PDU and terminate the connection event. (`Line 3741 in ble_ll_conn.c`), which can cause a larger transmission latency.
So, I think this can be easily improved. Specifically, the Central can still transmit an empty PDU to ask the Peripheral to immediately retransmit the packet.

andrzej-kaczmarek, 2024-11-25:
The connection event is terminated if one packet is lost or two consecutive packets are received with an invalid CRC (vol 6, part B, 4.5.6). We cannot reply to a packet that we did not receive.

sada45, 2024-11-25:
To my understanding, the `line 3741 in ble_ll_conn.c` shows the cases that the packet is received but with invalid CRC. If there is only one invalid CRC, the Central can still reply.
[image]

sada45, 2024-11-25:
The Line 3735 is correct. But the the question is in the `else` block.
If there is only one CRC error and the Centeral has no data to send `connsm->flags.last_txd_md=0`, the Central still don't reply the peripheral.
```
#if MYNEWT_VAL(BLE_LL_ROLE_CENTRAL)
            case BLE_LL_CONN_ROLE_CENTRAL:
                reply = connsm->flags.last_txd_md;
                break;
#endif
```

andrzej-kaczmarek, 2024-11-25:
Ok, now it's clear. This should fix the issue: https://github.com/apache/mynewt-nimble/pull/1926

closed 2025-12-09 by sjanc