apache/mynewt-nimble #1976: BLE app build documentation fails to build the configure and build the app when following the documentation guidelines

rabarar, 2025-02-16:
I’m new to newt/nimble (experienced embedded dev) and am trying to build the BLE example app from the Docs following STEP-BY-STEP instructions and getting the following when I attempt to build the BASIC BLE app:

editing the `apps/ble_app/syscfg.yml` file as specified in the docs produces this error when building
```
Building target targets/ble_tgt
Error: Override of defunct settings detected:
    BLE_HCI_TRANSPORT: see doc/transport.md

Setting history (newest -> oldest):
    BLE_HCI_TRANSPORT: [apps/ble_app:builtin, @apache-mynewt-nimble/nimble/transport:
```

moreover, if I take out the the  `BLE_HCI_TRANSPORT: builtin` in `apps/ble_app/syscfg.yml` and try to build, I get compile errors :

```
Error: repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_hci_supp_cmd.c:299:5: error: initializer element is not constant
     octet_0,
     ^
repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_hci_supp_cmd.c:299:5: note: (near initialization for 'g_ble_ll_hci_supp_cmds[0]')
repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_hci_supp_cmd.c:301:5: error: initializer element is not constant
     octet_2,
     ^
...
```

sjanc, 2025-02-17:
Hi,

I think tutorials might be slightly not up to date...

Just remove  BLE_HCI_TRANSPORT: builtin

rabarar, 2025-02-17:
Yes.  But part two of my message said that removing it allows the build to proceed, but fails to compile on non-static initializer errors for octet definitions…

sjanc, 2025-02-17:
yes, as mentioned it seems to build for me (from tutorials)

could you show output of 'newt info' as well as compiler version you use?

rabarar, 2025-02-17:
So, more to the point,

The repos/apache-mynewt-nimble/nimble/controller/src/ble_ll_hci_supp_cmd.c defines two macros 
`#define BIT(n)      (1 << (n)) |`
`#define OCTET(x)    (0 | x 0)`

Both look incorrect 

And then they’re used to initialize static const uint8_t g_ble_ll_hci_supp_cmds[64] 

But without adding an else case, for example in octet_0, (modified to add the #else 0)

```
static const uint8_t octet_0 = OCTET(
#if MYNEWT_VAL(BLE_LL_ROLE_PERIPHERAL) || MYNEWT_VAL(BLE_LL_ROLE_CENTRAL)
    BIT(5) /* HCI Disconnect */
#else
    0
#endif
);
```

Compilation fails on erroneous initializers. Thoughts on how this gets resolved? PR? With default values of 0?

sjanc, 2025-02-17:
Hi,

Empty OCTET macro expands to  (0 | 0) which looks like const value to me...
And BIT macro is just helper to be used inside OCTET macro.

What compiler do you use?

sjanc, 2025-02-17:
I just noticed that you mentioned gcc 5.4.1

TBH I'd really suggest to use something newer, I don't think we test with such old GCCs...

rabarar, 2025-02-17:
ah - I think I have an old `arm-none-eabi-gcc` lurking!  my `arm-none-eabi-gcc (GCC) 14.2.0` version isn't being used - let me relink and rebuild and see if that's the issue.

rabarar, 2025-02-17:
resolved:  OSX had a legacy arm toolchain lurking - removed, reinstall `gcc-arm-embedded` which contains the latest toolchain and successfully built the target - thanks!!

closed 2025-02-17 by sjanc