apache/mynewt-nimble #2004: Potentially blocking call within critical section

gmarull, 2025-03-07:
In the following controller code:

https://github.com/apache/mynewt-nimble/blob/7b14eab3d07caffa3a70046271b9f5b805f5c965/nimble/controller/src/ble_ll_rfmgmt.c#L285-L300

`ble_ll_event_add` is called while inside a critical section. This calls:

https://github.com/apache/mynewt-nimble/blob/7b14eab3d07caffa3a70046271b9f5b805f5c965/nimble/controller/src/ble_ll.c#L1426-L1430

In case of FreeRTOS,

https://github.com/apache/mynewt-nimble/blob/7b14eab3d07caffa3a70046271b9f5b805f5c965/porting/npl/freertos/include/nimble/nimble_npl_os.h#L100-L104

https://github.com/apache/mynewt-nimble/blob/7b14eab3d07caffa3a70046271b9f5b805f5c965/porting/npl/freertos/src/npl_os_freertos.c#L60-L80

As you can see above, `xQueueSendToBack` is called with `portMAX_DELAY`. I guess at the very least, we should add a zero delay if we're within a critical section. Also, see https://forums.freertos.org/t/critical-sections-freertos-api-calls/17352.

andrzej-kaczmarek, 2025-03-07:
I think we could try to remove critical sections from rfmgmt code entirely. The only usage of rfmgmt outside LL task context seems to be the rfmgmt timer, at least this is the only I could find after a quick scan though the code. If we change that to either send an event to LL queue in order to serialize rfmgmt calls or replace the timer with a callout, critical sections should not be necessary.

andrzej-kaczmarek, 2025-03-07:
Ah, I just realized that sending an event from timer callback is not great idea as this would delay rfmgmt enable a bit which is not what we want there...