apache/mynewt-nimble #2081: ble_hs_adv_find_field() may dereference invalid memory for malformed advertising data

kscheff, 2025-07-16:
https://github.com/apache/mynewt-nimble/blob/3bb26716e66f8db4663e7972d27fce9c09898fca/nimble/host/src/ble_hs_adv.c#L755

The function [ble_hs_adv_find_field()](https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_hs_adv.c) directly casts a buffer offset to struct ble_hs_adv_field * before verifying that the pointer is within bounds. This can lead to a crash (e.g., Load access fault) if the advertising data is malformed or truncated.

sjanc, 2025-07-24:
Hi,

There is both min length check (in while loop) so that `->length` can be check, and then later per fields length is validated.
I don't see any issues with accessing  past buffer length. Could you clarify on where the issue is?

sjanc, 2025-07-25:
Hi,

ad 1)  this whole block is in  `while (length > 1) ` so accessing length (and type) is always safe
ad 2)  same as above,  next iteration is protected by while loop condition
ad 3) pretty much the same,  both length and type are verified before access, not sure where this claim come from

Only issue I see in current implementation is that it would simply ignore leftover last byte without validating if this is terminator (ie zero),  but those wouldn't be valid LTV anyway. Stil, no invalid memory access here.

regarding recommended fix: this "fix" would introduce invalid memory access if field_len==255...,  and checking for `length < 1` inside `while (length > 0)` is dead code

PS
please cross-check your chatbot output before posting

kscheff, 2025-07-25:
Point is that it results in a panic - there is no discussion. 

The field is a 2 byte value. It seems to crash when the length is an odd number. So when accessing the last byte it crashes due to the field access.

Thanks for pointing out the difference in my implementation (however that return code it's not used in my app).