apache/mynewt-nimble #2095: Linux port mutex lock can silently fail

Digitelektro, 2025-07-31:
I have found two bugs in the Linux port implementation.

The first issue is that `BLE_NPL_TIME_FOREVER` is defined as `INT32_MAX`. I'm not sure if this is intentional, but after reviewing the code, using `UINT32_MAX` would make more sense. Several calls to `ble_npl_mutex_pend` use 0xFFFFFFFF, and the timeout parameter of `ble_npl_mutex_pend` is declared as uint32_t.
For example, `ble_hs_lock_nested` calls it with 0xFFFFFFFF. However, I also see some functions using `BLE_NPL_TIME_FOREVER`, so there is an inconsistency.

The more critical problem is that if `ble_npl_mutex_pend` ends up calling `pthread_mutex_timedlock`, it can fail, and the callers don't check the result. In such cases, `pthread_mutex_timedlock` sometimes returns `EINVAL`. This happens because `mu->wait.tv_nsec` cannot be greater than or equal to 1,000,000,000. As a result, the mutex is not locked, which leads to instability issues in multi-threaded application.

Possible solution:

```
ble_npl_error_t
ble_npl_mutex_pend(struct ble_npl_mutex *mu, uint32_t timeout)
{
    int err;

    if (!mu) {
        return BLE_NPL_INVALID_PARAM;
    }

    //here uint32_t compared to INT32_MAX
    if (timeout == BLE_NPL_TIME_FOREVER) { 
        err = pthread_mutex_lock(&mu->lock);
    } else {
        err = clock_gettime(CLOCK_REALTIME, &mu->wait);
        if (err) {
            return BLE_NPL_ERROR;
        }

        mu->wait.tv_sec  += timeout / 1000;
        mu->wait.tv_nsec += (timeout % 1000) * 1000000;

        // Limit nsec value to less than 1000 million
        if (mu->wait.tv_nsec >= 1000000000) {
            mu->wait.tv_sec += mu->wait.tv_nsec / 1000000000;
            mu->wait.tv_nsec = mu->wait.tv_nsec % 1000000000;
        }

        err = pthread_mutex_timedlock(&mu->lock, &mu->wait);
        if (err == ETIMEDOUT) {
            return BLE_NPL_TIMEOUT;
        }
    }

    return (err) ? BLE_NPL_ERROR : BLE_NPL_OK;
}
```

Digitelektro, 2025-07-31:
Thanks for confirming of the UINT32_MAX.

Can you point me out where it is limited to 999000000? 
I added a debug print to see what happens when `pthread_mutex_timedlock` fails:

```
        err = pthread_mutex_timedlock(&mu->lock, &mu->wait);
        if(err) {
            printf("Mutex lock failed, err=%d, sec=%d, nsec=%d\n", err, mu->wait.tv_sec, mu->wait.tv_nsec);
        }
```

This is what I get, it is definitely over 1000 million:

 _Mutex lock failed, err=22, sec=1758269008, nsec=1039810503_

Adding the code I recommended in my first post, this never fails again.

sjanc, 2025-08-06:
Hi,

Should be fixed by https://github.com/apache/mynewt-nimble/pull/2101

Digitelektro, 2025-08-06:
Thank you so much your quick response and fix! And sorry for not making a pull request, I just didn't have time yet!

sjanc, 2025-08-14:
should be fixed in master